libretro/Lakka-LibreELEC #1668: [Feature request] Support Wi-Fi on Orange Pi Plus 2E (Realtek RTL8189ETV, IEEE 802.11 b/g/n)

d-apps, 2022-03-29:
Is it possible to support Wi-Fi in Orange Pi Plus 2E (Realtek RTL8189ETV, IEEE 802.11 b/g/n) please?

Thank you.

ToKe79, 2022-03-30:
Hi,
based on information from [here](https://linux-sunxi.org/Wifi#:~:text=%C2%A0%3F%3F-,RTL8189ES%20/%20RTL8189ETV,make%20%2Dj4%20ARCH%3Darm%20CROSS_COMPILE%3Darm%2Dlinux%2Dgnu%2D%20KSRC%3D../linux,-RTL8189FTV) I added the driver/module. You can test with image from [here](https://nightly.builds.lakka.tv/members/vudiq/h3-rtl8189es/) and give me feedback. If positive, then it can be included in the next version.

d-apps, 2022-04-01:
Hi,
Is there any way a can keep all the configurations I made in my current image installed, before installing this image? I have around 20gb of roms and lots of custom configs.

Thanks

d-apps, 2022-04-01:
@ToKe79 I just tested and it didn't work.

I enabled Wi-Fi, it scans, after scan complete it doesn't show any network.

[image]

ToKe79, 2022-04-03:
please, provide some logs from the system.
[Troubleshooting Guide](http://www.lakka.tv/doc/Troubleshooting-Lakka/)

d-apps, 2022-04-29:
> please, provide some logs from the system. [Troubleshooting Guide](http://www.lakka.tv/doc/Troubleshooting-Lakka/)

Sorry for the delay.
Logs were uploaded.

[log-2022-04-29-13.50.57.zip](https://github.com/libretro/Lakka-LibreELEC/files/8591289/log-2022-04-29-13.50.57.zip)

Ntemis, 2022-06-10:
I checked and even after the driver is enabled there is no wifi in the system logs so the board file might need fixing + no one of the team owns this device so no way to fix it

closed 2022-06-10 by Ntemis