libretro/Lakka-LibreELEC #1669: lakka4.0 x86 bug report

greatliu, 2022-03-29:
found bug:
1.lakka system cannot boot in nvidia card. I used it corrently in amd card computer;
2.cannot scan(import content) wbfs file for wii, I can ues wbfs only by "load content"; 
3.cannot select GC controllor for wii games, e.g. mario cart wii.

Thanks.

ToKe79, 2022-03-29:
Hi,
Re issue 1: is fixed in dev/nightly build
Re issue 2: format is probably not supported by libretro database - @RobLoach ?
Re issue 3: did it work in Lakka 3.7?

greatliu, 2022-04-01:
> Hi, Re issue 1: is fixed in dev/nightly build Re issue 2: format is probably not supported by libretro database - @RobLoach ? Re issue 3: did it work in Lakka 3.7?
Hi, my friend.
1. Thanks, I am waiting for lakka 4.1 release;
2. I can scan wbfs in previous version, the bug only exists in 4.0;
3. before lakka 4.0, all the dophine emu will crash on my pc(HP elite 8300usdt, i5 3470s, 8g ram, AMD firepro m5100), so 4.0 is the first version I can run wii game. I wish to emulate GC gamepad using my xbox360 controllor.

greatliu, 2022-04-01:
> 2. Use Manual Scan for content that won't scan. Wbfs serial scanning isn't supported in RA right now.

Hi,
But I can scan wbfs in lakka 3.x.
Thanks.

ToKe79, 2022-04-01:
> 3. before lakka 4.0, all the dophine emu will crash on my pc(HP elite 8300usdt, i5 3470s, 8g ram, AMD firepro m5100), so 4.0 is the first version I can run wii game. I wish to emulate GC gamepad using my xbox360 controllor.

so the issue is that your xbox360 controller is not working in Lakka at all, or that the mapping is not correct for the Wii?

RobLoach, 2022-04-01:
> But I can scan wbfs in lakka 3.x.

Interesting. WBFS scanning code didn't change, I don't think. Made an issue over at https://github.com/libretro/RetroArch/issues/13810

greatliu, 2022-04-02:
I mean, I can not select GC controllor in wii game "controllor" setting, the aviliable selection is wiimote, wiimote(sideways) and so on.

As I know, a lot of wii game support GC controllor, like resident evil, mario kart wii......

***@***.***
 
From: Tomáš Kelemen
Date: 2022-04-01 18:14
To: libretro/Lakka-LibreELEC
CC: greatliu; Author
Subject: Re: [libretro/Lakka-LibreELEC] lakka4.0 x86 bug report (Issue #1669)
before lakka 4.0, all the dophine emu will crash on my pc(HP elite 8300usdt, i5 3470s, 8g ram, AMD firepro m5100), so 4.0 is the first version I can run wii game. I wish to emulate GC gamepad using my xbox360 controllor.
so the issue is that your xbox360 controller is not working in Lakka at all, or that the mapping is not correct for the Wii?

greatliu, 2022-04-02:
> > But I can scan wbfs in lakka 3.x.
> 
> Interesting. WBFS scanning code didn't change, I don't think. Made an issue over at [libretro/RetroArch#13810](https://github.com/libretro/RetroArch/issues/13810)

Thanks for support. I cannot understand the code. I will waiting for next release.

RobLoach, 2022-04-03:
2. Pkos found a an update for WBFS scanning which is now in RetroArch `master`. Hopefully the additional magic values help :+1:

greatliu, 2022-04-03:
thanks, my friend.

***@***.***
 
From: Rob Loach
Date: 2022-04-03 10:54
To: libretro/Lakka-LibreELEC
CC: greatliu; Author
Subject: Re: [libretro/Lakka-LibreELEC] lakka4.0 x86 bug report (Issue #1669)
Pkos found a an update for WBFS scanning which is now in RetroArch master. Hopefully the additional magic values help

ToKe79, 2022-04-03:
to your 3rd issue - gamecube controller - I just tried one wii iso, after you start the content, open the RetroArch menu (F1 key or your mapped button combination), scroll down to `Controls`, then `Port 1 Controls`, then `Device Type` and the list should show:

- `None`
- `WiiMote`
- `WiiMote (sideways)`
- `WiiMote + Nunchuk`
- `WiiMote + Classic Controller`
- `WiiMote + Classic Controller Pro`
- `Real WiiMote`
- `GameCube Controller`

This was tested on latest nightly (on par with 4.1-rc, which also solves your Nvidia issue).

ToKe79, 2022-04-06:
wbfs scanning with the patch from @pkos works.

@greatliu did you manage to use the `GameCube Controller`?

All reported bugs should be fixed now in the Lakka-v4.x branch. Closing the issue

closed 2022-04-06 by ToKe79

greatliu, 2022-04-08:
Ver 4.1RC cannot scan WBFS.
I will wait next release.
Thanks all the same for your work.

***@***.***
 
From: Tomáš Kelemen
Date: 2022-04-07 02:04
To: libretro/Lakka-LibreELEC
CC: greatliu; Mention
Subject: Re: [libretro/Lakka-LibreELEC] lakka4.0 x86 bug report (Issue #1669)
wbfs scanning with the patch from @pkos works.
@greatliu did you manage to use the GameCube Controller?
All reported bugs should be fixed now in the Lakka-v4.x branch. Closing the issue