libretro/Lakka-LibreELEC #1673: Lakka 4.x RK3328.aarch64 random black screen

yikesmeguts, 2022-04-01:
I have noticed that since the 4.x branch was released, the video on my Rock64 (RK3328) will randomly fail to display video either: 

- upon launching cores
- upon returning to the Retroarch menu after exiting a core
- sometimes upon initial boot

When the video fails to display, it is clear that the core has not crashed.  Audio still plays in the background, and games will play, just without displaying content to the screen.  This behavior was not present in the 3.x .arm builds.  I have observed this behavior after upgrading from Lakka 3.7, and also after a clean install of 4.x.  The behavior does not appear to be related to any particular core or rom, sometimes exiting and restarting a successfully launched core will cause the video to fail to display.

I have not observed this problem on any other platform that I have tested, only the Rock64/RK3328.
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/8397250/log.txt)

**Which version of Lakka are you using?**  Lakka 4.0 or any 4.x nightly RK3328.aarch64, up to and including the 1 April 2022 nightly
**What system hardware are you using?**  RK3328 Rock64 with 1 Gb RAM, 64 Gb eMMC
**What did you do?** Started the device, launched various cores.
**What did you expect to happen?** Video to display consistently.
**What happened instead?** Video will randomly fail to display, but I have been unable to detect any pattern of behavior.

gouchi, 2022-04-01:
It might coming from the switch to open source drivers [lima](https://docs.mesa3d.org/drivers/lima.html) and [panfrost](https://docs.mesa3d.org/drivers/panfrost.html).

You are using `lima` driver on RK3328.

yikesmeguts, 2022-04-01:
That seems to be the case; the changes between 3.x and 4.x are far more numerous than just updating the version number. :^)

vudiq has a test build underway, and I’ll try it out as soon as the build is ready. If the open source drivers eventually fully support the Rock64’s gpu, great! If not, I can always revert back to 3.7.

Thanks to all for their time and efforts.

Ntemis, 2022-06-10:
Thank you for the feedback

closed 2022-06-10 by Ntemis