libretro/Lakka-LibreELEC #1675: MAME cv1k game Not working at all.

vus7294, 2022-04-04:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 4.1 test ver.
- What system hardware are you using? Nintendo Switch
- What did you do? Booting any of CV1K game(7z) Mushihimesama or ESPGaluda 2, Daifukkatsu and many other
- What did you expect to happen? Boot Normally
- What happened instead? Crash and force Reboot Lakka

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

I Tested by Now. Same as MAME 2015. so That Cores seem very Unstable. but  It seems work with FBNeo's Games.

gouchi, 2022-04-06:
I tried on x86_64 using [latest nightly](https://nightly.builds.lakka.tv/latest/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20220406-62399ee.img.gz) and it is working so maybe an issue on arm/aarch64 ?

gouchi, 2022-04-07:
I tested also on arm and it is working so maybe related only to Switch aarch64 ?

vus7294, 2022-04-08:
[ibarablk.log](https://github.com/libretro/Lakka-LibreELEC/files/8448304/ibarablk.log)
i got logfile. crashed at last line

vus7294, 2022-04-08:
[mame2015.log](https://github.com/libretro/Lakka-LibreELEC/files/8448859/mame2015.log)
[mame2022.log](https://github.com/libretro/Lakka-LibreELEC/files/8448860/mame2022.log)
gdb backed up log here.

gouchi, 2022-04-08:
Hi,

Thank you for the log.

Did you try with video driver `glcore` if you are having  also the issue ?

Kind regards,

vus7294, 2022-04-10:
Same result. ;(

Ntemis, 2022-06-10:
Vulkan is not supported on mame or fbneo cores switch to gl

closed 2022-06-10 by Ntemis