libretro/Lakka-LibreELEC #1676: Some wishes, about bluetooth and ps2.

greatliu, 2022-04-08:
I am a fan of lakka, I use lakka for about 2 years.
Sometimes I use batoce.. instead. I found something need to be improved in lakka.
1. bluetooth connect. In lakka UI, I connect xbox one controllor or wii mote, it can be connected just 1 second and then disconnected.
2. can you add xbox one wireless adapter driver for lakka?
3. before version 4.0, wii(gc) and ps2 emu always crash. After 4.0 wii is ok, but ps2 still keep crash(if you boot ps2 game, it just jumps back to lakka main menu). My PC is hp8300usdt(i5 3470s, 8g, firpro m4000m).
4. can you add other font support like arial?

I like lakka verymuch so I wish it become perfect. Thanks for all the developpers.

gouchi, 2022-04-09:
Hi,

>1. bluetooth connect. In lakka UI, I connect xbox one controllor or wii mote, it can be connected just 1 second and then disconnected.

It should work. You should try to provide some log as written in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/.github/ISSUE_TEMPLATE.md).

>2. can you add xbox one wireless adapter driver for lakka?

It is already [added](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/packages/linux-drivers/xpadneo/package.mk). You should check the [troubleshooting section](https://atar-axis.github.io/xpadneo/#troubleshooting) and also the [compatible BT dongles](https://atar-axis.github.io/xpadneo/#bt-dongles) for the best result.

>3. before version 4.0, wii(gc) and ps2 emu always crash. After 4.0 wii is ok, but ps2 still keep crash(if you boot ps2 game, it just >jumps back to lakka main menu). My PC is hp8300usdt(i5 3470s, 8g, firpro m4000m).

Same please provide some log and it will depend which core you are using [Pcsx2](https://docs.libretro.com/library/pcsx2) or [Play!](https://docs.libretro.com/library/play/).

>4. can you add other font support like arial?

You can add them to Lakka then load it with Settings tab  > User Interface > Appearance > Menu font if you are using XMB menu.

Thank you.

greatliu, 2022-04-11:
Hi,
Question 1.2.&4
Thanks for your help, I will have a try.

Question 3. I found in 4.1 release PS2 game runs!

My question is still no.2, I mean can you add !official! xbox one wireless adapter driver, not the bluetooth driver.

Thanks, my friend.

***@***.***
 
From: gouchi
Date: 2022-04-10 04:39
To: libretro/Lakka-LibreELEC
CC: greatliu; Author
Subject: Re: [libretro/Lakka-LibreELEC] Some wishes, about bluetooth and ps2. (Issue #1676)
Hi,
bluetooth connect. In lakka UI, I connect xbox one controllor or wii mote, it can be connected just 1 second and then disconnected.
It should work. You should try to provide some log as written in the issue template.
can you add xbox one wireless adapter driver for lakka?
It is already added. You should check the troubleshooting section and also the compatible BT dongles for the best result.
before version 4.0, wii(gc) and ps2 emu always crash. After 4.0 wii is ok, but ps2 still keep crash(if you boot ps2 game, it just >jumps back to lakka main menu). My PC is hp8300usdt(i5 3470s, 8g, firpro m4000m).
Same please provide some log and it will depend which core you are using Pcsx2 or Play!.
can you add other font support like arial?
You can add them to Lakka then load it with Settings tab > User Interface > Appearance > Menu font if you are using XMB menu.
Thank you.

ToKe79, 2022-04-11:
Can you be more specific, what do you mean by the official driver? Especially where to get it. I am aware of opensource [`xone` driver](https://github.com/medusalix/xone) - you mean this one?

greatliu, 2022-04-12:
> Can you be more specific, what do you mean by the official driver? Especially where to get it. I am aware of opensource [`xone` driver](https://github.com/medusalix/xone) - you mean this one?

I mean, the support for "xbox wireless adapter", the official usb wireless adapter for xbox one controller.
In the previous lakka version, the adapter did not work.

Thanks.

Ntemis, 2022-06-10:
Your feedback will be taken into consideration,feel welcome to come over to our discord so we can help you more interactive

closed 2022-06-10 by Ntemis