libretro/Lakka-LibreELEC #1678: Lakka 4.1 bootloop

badarg1, 2022-04-10:
I have 2 old laptops I would like Lakka on. They are very similar and suffer the same issue: they are stuck in the "flower bootloop". I'm using Lakka 4.1. I didn't try a nightly build as this version was released just 2 days ago (and the issue is documented in the FAQ, so I don't thik it has been fixed since then).

Here are the slight behavious differences I've noticed between the 2 laptops and their logs (I believe hardware specs are already in the logs):

**Toshiba NB200**
The flower/berry in the logo is shown with a green stem and leaves and orange/yellow "balls".
The prompt in text mode is green.
Logs: [nb200.zip](https://github.com/libretro/Lakka-LibreELEC/files/8458748/nb200.zip)

**Toshiba NB250**
The flower/berry in the logo is shown with a white stem and leaves and yellow "balls".
The prompt in text mode is white.
Logs: [nb250.zip](https://github.com/libretro/Lakka-LibreELEC/files/8458749/nb250.zip)

gouchi, 2022-04-10:
As the [FAQ mentioned](https://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow), did you try to disable menu shader pipeline ? You will have to boot to [textmode](https://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) to edit the [configuration file](https://www.lakka.tv/doc/Configuring-Lakka/).

Your GPU is maybe too old for the menu shader.

If it is still not working you will have to switch to RGUI menu driver.

badarg1, 2022-04-10:
Thanks for the reply.
Neither disabling the menu shader pipeline nor switching the menu driver to RUI made any differente.

I'm not sure I'm doing the latter correctly, though. The [documentation](https://www.lakka.tv/doc/Menu-drivers/) only tells how to change the menu driver from the GUI, which is not viable in my case. What I did was change the option `menu_shader` to `"rgui"` in the configuration file. I'm not sure this is correct as the default value was `"ozone"`, which is not in the documentation.

gouchi, 2022-04-10:
You  should have in the configuration file 

```
menu_shader_pipeline = "0"
menu_driver   = "xmb"
```
and for rgui menu driver

`menu_driver = "rgui"`

badarg1, 2022-04-10:
That's what I have (with `"rgui"`). Still stuck in the bootloop.

In my previous message I meant `menu_driver`, not `menu_shader`, sorry.

Ntemis, 2022-06-10:
Unsupported gpu driver

closed 2022-06-10 by Ntemis