libretro/Lakka-LibreELEC #1679: No network on odroid c2 since Lakka 4.0

aeloya, 2022-04-10:
There is no network configuration for eth0 in Lakka 4.0 and 4.1. 
Last system message on screen tells that eth0 cannot be configured. 
Lakka 3.7 works fine.

Ntemis, 2022-06-10:
Odroid c2 works fine here, hw issue maybe?

closed 2022-06-10 by Ntemis

dehart, 2022-08-12:
@Ntemis I don't think it's a hardware issue. I have the exact same experience.

I've made some shots of the log. Hope it's useful

[image]
[image]
[image]