libretro/Lakka-LibreELEC #1680: Enable snes9x 2005 for rpi0/GPICase

smilerliu, 2022-04-11:
RG350 has a 1GHz MIPS CPU and it runs snes9x 2005 pretty well. Can we include the core in the rpi0/GPICase build as well? I'll be happy to at least try it out and report back how the core works.

ToKe79, 2022-04-14:
Hi @smilerliu 
you can test cores (`snes9x2005` and `snes9x2005_plus`) from [here](https://nightly.builds.lakka.tv/members/lakkabuildbot/cores/) - let us know how they perform.
Thank you!

smilerliu, 2022-04-15:
I only have a GPI case setup and I don't have the vanilla setup for running the rpi1 build.

I tried a few games (each for a few minutes) with the 2005 core, with the default config:

- Crono Trigger: mostly full speed with frameskip off.
- Mega Man X: full speed with frameskip off.
- Street of Fighter II Turbo: full speed with frameskip off.
- Super Mario RPG: slowdown and choppy even with frameskip set to auto; barely playable (to my standard).
- Zelda a Link to the Past: framerate constantly around 55 FPS with frameskip off; the gameplay feels normal though (again, to my standard).

In my perception, the core performs slightly worse than the 2002. I heard that the 2005 core has much better game compatibility than the 2002 core; I think it's worth adding to the GPI case build.

The 2005 plus core is much slower and I wouldn't want it to be in the GPI case build (in my opinion).

smilerliu, 2022-04-15:
BTW the 2005 core performs much worse compared to RetroArch on RG350. There it plays all the games Iisted above at full speed. Perhaps this is an expected difference between the two platforms (although both have a 1GHz single core CPU).

closed 2022-04-15 by ToKe79

smilerliu, 2022-04-17:
For the record, I tried a few more games. For all the games I tried, the only one that doesn't work well with snes9x2002 is the Mario RPG game. Some people have reported that Super Metroid didn't work well with snes9x2002, but when I try now, it actually works pretty well. So it seems the core is still being improved, and I haven't been able to see the value of the snes9x2005 core yet.

I would be great if someone can point out a case when snes9x2005 runs better than snes9x2002 on an GPI case + Zero W 1.

commit 529f0bb3 (2022-04-15): RPi/GPICase: include snes9X2005 core

closes #1680