libretro/Lakka-LibreELEC #1682: Failed to mount

nelson80s, 2022-04-13:
Good morning, I don't know if this is the right place, please correct me. Does anyone have any idea what it is or could help? 
My preference for Lakka for both PC and Arm S905x is because it is light, fast and well designed.
thanks
[image]
- -Which version of Lakka are you using?
- 4.1
- - What system hardware are you using?
- Amlogic S905x
- - What did you do?
I'm trying to install the new version 4.1 on a s905x box  I tried with several .dtb (p212 and etc) 
- -What did you expect to happen?
-The installation and the normal start of retroarch/lakka
- -What happened instead?
- the screen during boot looks like this according to the print. Freezes at failed to mount directories

ToKe79, 2022-04-21:
during the very first boot, did you get messages regarding expanding storage partition? can you try flashing the image again? what exact TV box do you have (not only the SOC, but manufacturer/model)?

nelson80s, 2022-04-29:
> during the very first boot, did you get messages regarding expanding storage partition? can you try flashing the image again? what exact TV box do you have (not only the SOC, but manufacturer/model)?

Sorry by delayed Feedback
Im newbie on these Technologies and github
Yes. The first boot was ok with expandings etc
This issue occurs at the second boot.
I will try with new lakka 4.2 
The tv box is a super console x pro 1gb p212
Tv is a smart lg 32 in 

Cheers

Ntemis, 2022-06-10:
Nobody from the team owns a console x pro  and looks like meson-gxl-s905x-p212.dtb is incompatible with your board,
tough luck

closed 2022-06-10 by Ntemis