libretro/Lakka-LibreELEC #1686: Orange Pi 4 LTS support

soda-pop-ice-cream, 2022-04-20:
Hello!
It's possible to add support of Orange Pi 4 LTS board? It's same Orange Pi 4, but with diffirent WiFi + BT chip(CDW 20U5622-00 instead of AP6256)
On Lakka website listed literally all RK3399 boards except it. I tested **all** provided RK3399 boards images with **no** luck(and sorry for generating some network load).
I **had** luck booting up image for ROCKPro64 board from Batocera, and everything works except WiFi and BT.
I have UART and would be glad to test images and provide logs.

[image]
.

slowpokefarm, 2022-05-25:
+1 to this. Original Orange Pi 4 seems not to be even available for purchase anymore with only LTS version available.

Ntemis, 2022-06-10:
Unfortunately orange pi dont send R&D boards to our team so we cant fix this

slowpokefarm, 2022-06-27:
I have contacted Orange Pi customer support about this issue with Pi 4 LTS drivers and they responded positively about them willing to help to resolve it. They also mentioned they have contacted Lakka team and @Ntemis specifically. Please tell if there is some progress with this issue in case they actually did contact anyone.
I kind of already regret getting LTS boards for Lakka.

Ntemis, 2022-06-27:
@slowpokefarm can you come over to lakka channel over discord/retroarch ?
I don't remember been ever contacted by them for the Orange Pi 4 or any board

dmrlawson, 2022-07-21:
@slowpokefarm @soda-pop-ice-cream please could you have a go with this image?

https://nightly.builds.lakka.tv/members/dmrlawson/Lakka-RK3399.aarch64-4.2-devel-20220717183455-62320b3-orangepi-4-lts.img.gz

I expect that sound/WiFi/ethernet/Bluetooth do not work yet, but other things should

soda-pop-ice-cream, 2022-07-22:
@dmrlawson thank you! I tested it and results so far:
Sound - work!
WIFI - doesn't work
Bluetooth -  doesn't work
Ethernet - ethernet port itself works, it has link, but can't acquire IP-addres with DHCP(dispaying 169.254.*.* IP-address). I tried adding some random ip from my router subnet as static via connman but sadly it didn't work too, even after i manually added default route

So, i tested some random snes, nes, megadrive games, even with shaders, and everything emulation-related works. Some logs:

[dmesg.log](https://github.com/libretro/Lakka-LibreELEC/files/9164319/dmesg.log)
[boot.log](https://github.com/libretro/Lakka-LibreELEC/files/9164315/boot.log)
[connman.log](https://github.com/libretro/Lakka-LibreELEC/files/9164316/connman.log)
[log-2021-10-29-09.24.32.zip](https://github.com/libretro/Lakka-LibreELEC/files/9164321/log-2021-10-29-09.24.32.zip)

slowpokefarm, 2022-09-01:
I have tested the latest available image on nightly list, the results are similar to those of @soda-pop-ice-cream

Networking doesn't work, neither wireless nor wired. eth0 interface receives wrong IP address, network doesn't work after setting a valid address manually.

Bluetooth doesn't work.

dmrlawson, 2022-10-17:
@slowpokefarm @soda-pop-ice-cream please would you try out this build?

https://nightly.builds.lakka.tv/members/dmrlawson/Lakka-RK3399.aarch64-5.0-devel-20221016235533-cc0fbe3-orangepi-4-lts.img.gz

Ethernet is working for me now, I'll try to get the wifi/bt going next.

I get a black screen with mupen64plus. Dolphin works but is slow.

soda-pop-ice-cream, 2022-10-18:
@dmrlawson doesn't boot for me:
```
U-Boot TPL 2022.10 (Oct 16 2022 - 23:39:53)                                  
sdram_init: DDR3 - 666MHz failed!                                            
rk3399_dmc_init DRAM init failed -22                                         
Missing DTB
```

dmrlawson, 2022-10-18:
> @dmrlawson doesn't boot for me:
> 
> ```
> U-Boot TPL 2022.10 (Oct 16 2022 - 23:39:53)                                  
> sdram_init: DDR3 - 666MHz failed!                                            
> rk3399_dmc_init DRAM init failed -22                                         
> Missing DTB
> ```

Hm, sorry about that. Maybe I linked the wrong version. Will check tonight.

dmrlawson, 2022-10-18:
@soda-pop-ice-cream I just flashed the image at the link I posted above to a fresh SD card and it worked. Would you mind trying again?

soda-pop-ice-cream, 2022-10-19:
Well, I tried 2 different SD cards, had no luck on both. Previous build that you provided in july boots just fine on both of them.

dmrlawson, 2022-10-19:
Very strange. Thank you very much for trying again. 

The only thing I can think of at the moment is to get more data. @Ntemis did you have one of these?

Ntemis, 2022-10-19:
I do. I can test tonight

dmrlawson, 2022-10-23:
@soda-pop-ice-cream I suspect one of two things:
 - Somehow the image didn't flash properly. What process are you using?
 - Your emmc has changed somehow, going by this thread: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=4982
   - Have you installed armbian to the internal storage at some point?
   - It sounds like even when that's the case the official images can still boot from sd. I guess we'd need to figure out what's different :-|
 
Could you try the older image you tried before and reported working? If this no longer works I think it's option two. If it does and the new image doesn't work when flashed with the same process I've no idea what's going on.

slowpokefarm, 2022-12-20:
Is there any progress on this? I can't test the latest image at the moment since I'm away from hardware, just wanted to know if anyone tested it since October.

dmrlawson, 2022-12-20:
> Is there any progress on this? I can't test the latest image at the moment since I'm away from hardware, just wanted to know if anyone tested it since October.

The patches to make it build with ethernet support are merged into the `Lakka-v5.x` branch. It boots for Ntemis and me but not soda-pop-ice-cream for some unknown reason.

I recently managed to build the wifi/bt drivers (or at least part of them) but haven't had any success getting them to work yet.

We don't have nightly builds for v5 yet but when we do I'll make sure we're building for it.

slowpokefarm, 2023-01-31:
I'm finally able to test the build, which image should I try?

ToKe79, 2023-01-31:
you can check [latest devel build](https://nightly.builds.lakka.tv/devbuild_latest/RK3399.aarch64) with `orange-pi4-lts` in the file name.

slowpokefarm, 2023-02-02:
> you can check [latest devel build](https://nightly.builds.lakka.tv/devbuild_latest/RK3399.aarch64) with `orange-pi4-lts` in the file name.

Ethernet works, no wifi/bluetooth, mupen64plus and PPSSPP show black screen only. Do I need to test anything else?

slowpokefarm, 2023-04-16:
Did some more testing with 22030304 build, I was able to run Dolphin with GameCube ROM exactly once and it was running smoothly without any issues or FPS drops. After that one time it always ended with black screen again with bad_ps_OGL_0.txt file created in Dump containing the following:
<details><summary>Details</summary>
<p>

#version 310 es

#define FORCE_EARLY_Z layout(early_fragment_tests) in

#define ATTRIBUTE_LOCATION(x)
#define FRAGMENT_OUTPUT_LOCATION(x)
#define FRAGMENT_OUTPUT_LOCATION_INDEXED(x, y)
#define UBO_BINDING(packing, x) layout(packing, binding = x)
#define SAMPLER_BINDING(x) layout(binding = x)
#define TEXEL_BUFFER_BINDING(x) layout(binding = x)
#define SSBO_BINDING(x) layout(binding = x)
#define IMAGE_BINDING(format, x) layout(format, binding = x)

#define VARYING_LOCATION(x)

#extension GL_OES_texture_buffer : enable
#extension GL_EXT_blend_func_extended : enable

#extension GL_EXT_shader_framebuffer_fetch: enable
#define FB_FETCH_VALUE real_ocol0
#define FRAGMENT_INOUT inout

precision highp float;
precision highp int;
precision highp sampler2DArray;
precision highp usamplerBuffer;
precision highp sampler2DMS;
precision highp image2DArray;
#define API_OPENGL 1
#define float2 vec2
#define float3 vec3
#define float4 vec4
#define uint2 uvec2
#define uint3 uvec3
#define uint4 uvec4
#define int2 ivec2
#define int3 ivec3
#define int4 ivec4
#define frac fract
#define lerp mix
SAMPLER_BINDING(0) uniform sampler2DMSArray samp0;
VARYING_LOCATION(0) in float3 v_tex0;
FRAGMENT_OUTPUT_LOCATION(0) out float ocol0;

void main()
{
  int layer = int(v_tex0.z);
  int3 coords = int3(int2(gl_FragCoord.xy), layer);
  ocol0 = texelFetch(samp0, coords, 0).r;
  for (int i = 1; i < 4; i++)
    ocol0 = min(ocol0, texelFetch(samp0, coords, i).r);
}
0:49(29): error: illegal use of reserved word `sampler2DMSArray'
0:49(29): error: syntax error, unexpected ERROR_TOK
\00
Dolphin Version: Dolphin [HEAD] 2f4b0f7
Video Backend: OpenGL ES

</p>
</details> 

I also was able to run a lot of more lightweight cores for NES, SNES, GBA etc without any issues.
I also was able to run PPSSPP core with x2 resolution without any fps drops or other issues.
I also tried to run Mupen64Plus-next core with N64 roms but each time Lakka was restarting without leaving any logs.

slowpokefarm, 2023-04-18:
Small update - I switched off MSAA for dolphin and it works with orange pi 4 lts under lakka devbuild v5.x.
Mupen64Plus-next does not work both on pi 3 lts and pi 4 lts. Always restarts lakka without producing any logs or dumps.

Ethernet works, wi-fi and bluetooth don't work.

gordaogameplay, 2024-02-05:
Hello, it's been a while since the topic had any related progress, I've had an orangepi 4 LTS board for a long time, and this month I started using it again but wanting to use the emulation resources. When searching on the internet I found Lakka, but their website does not have a specific image for orangepi 4 lts, so I found this topic. Has anyone had any progress on a fully functional image?

Ntemis, 2024-02-05:
https://nightly.builds.lakka.tv/2024-02-02_5.x/RK3399.aarch64/Lakka-RK3399.aarch64-devbuild-v5.x-20240202-0acbf0c-orangepi-4-lts.img.gz

gordaogameplay, 2024-02-05:
> https://nightly.builds.lakka.tv/2024-02-02_5.x/RK3399.aarch64/Lakka-RK3399.aarch64-devbuild-v5.x-20240202-0acbf0c-orangepi-4-lts.img.gz

Thanks @Ntemis, But, is it normal for the screen to go black and the Orange Pi to not show any blinking LEDs, indicating that something is being done by the system?

gordaogameplay, 2024-02-05:
> https://nightly.builds.lakka.tv/2024-02-02_5.x/RK3399.aarch64/Lakka-RK3399.aarch64-devbuild-v5.x-20240202-0acbf0c-orangepi-4-lts.img.gz

this img still not working, nothing happens when i turn on my orangepi 4 with th SD card =( any solution?

Ntemis, 2024-02-06:
will test this when i find some time

gordaogameplay, 2024-02-06:
> will test this when i find some time

Thanks a lot, i appreciate so much!

gouchi, 2024-04-09:
What is the status on the port ?

Can you make a test with [latest nightly](https://nightly.builds.lakka.tv/devbuild_latest/RK3399.aarch64/) ?

Thank you.

WiseToad, 2024-06-08:
Hello!

I tested my `Orange Pi 4 LTS` board with some versions of Lakka and got the following results:
- [Lakka-RK3399.aarch64-5.0-orangepi-4-lts.img.gz](https://le-builds.lakka.tv/RK3399.aarch64/Lakka-RK3399.aarch64-5.0-orangepi-4-lts.img.gz) *(stable 5.0)* - **FAILED** (the screen is blank after sbc power on, the on-board led does not blink)
- [Lakka-RK3399.aarch64-5.x-20240605-b411e54-orangepi.img.gz](https://nightly.builds.lakka.tv/5.x/2024-06-05/RK3399.aarch64/Lakka-RK3399.aarch64-5.x-20240605-b411e54-orangepi.img.gz) *(nightly 5.x)* - **FAILED** (same symptoms)
- [Lakka-RK3399.aarch64-6.x-20240603-a6eecbc-orangepi.img.gz](https://nightly.builds.lakka.tv/6.x/2024-06-03/RK3399.aarch64/Lakka-RK3399.aarch64-6.x-20240603-a6eecbc-orangepi.img.gz) *(nightly 6.x)* - **FAILED** (same symptoms)
- [Lakka-RK3399.aarch64-4.3-orangepi-4-lts.img.gz](https://le-builds.lakka.tv/RK3399.aarch64/Lakka-RK3399.aarch64-4.3-orangepi-4-lts.img.gz) *(stable 4.3)* - **OK** (screen starts to show logo and resize log, the on-board led starts flashing)

I noticed that the dtb file appeared on sd card after flushing 4.3 version of Lakka differs from the same file for the 5.0 version. So, maybe new version contains invalid dtb file and that the reason why new version doesn't work for me?

btw, I managed to build 5.x from source (it was challenge for me, the process wasn't smooth). So, result is the same - no image on screen, no flashing light after sbc power on with image built by me.

gordaogameplay, 2024-06-09:
Has anyone tested the versions shown in the post above to see if they have a 3DS emulator?

Ntemis, 2024-07-10:
OP request was fulfilled, i tested and is fully working on me, closing down.
Please open a new one if issues continue on next stable.

closed 2024-07-10 by Ntemis

gordaogameplay, 2024-07-14:
> OP request was fulfilled, i tested and is fully working on me, closing down. Please open a new one if issues continue on next stable.

What version are you using? Do you have an OrangePi 4 LTS or are you emulating? I don't mean to offend, because you clearly have much more knowledge than me, and I respect that, but I went to this link https://nightly.builds.lakka.tv , and downloaded literally every version from this site. I tested them all and none of them work, and it's not because I'm stupid, because all the Linux and Android versions I tested worked (not for emulation purposes), only the Lakka versions (all of them) don't work... and I apologize for the delay in testing and responding, but I work a lot and have little time to test, I'll be on vacation soon and I'll have time to try again. But this weekend I tested literally every build, date by date of modification of the builds, and none of them worked, whether 5.x or 6.x or 4.x .....