libretro/Lakka-LibreELEC #1688: Intel UHD 750 GPU isn't detected - Couldn't find a suitable DRM device.

gyllegert, 2022-04-20:
Using Lakka 4.1 and the latest nightly (2022-04-20) to test.

The system is a Lenovo ThinkCentre M70q Gen2
i5-10500t
8GB RAM in single channel configuration
256GB NVME SSD

I had a look at the log and it ends like this i both 4.1 and the latest nightly build.
RetroArch [ERROR] :: [KMS]: Couldn't find a suitable DRM device.
RetroArch [INFO] :: Found GL context: null
RetroArch [INFO] :: Detecting screen resolution 320x240.
RetroArch [INFO] :: [GL]: Vendor: (null), Renderer: (null).
RetroArch [INFO] :: [GL]: Version: (null).

ToKe79, 2022-04-21:
does your laptop have multiple display outputs? can you try disabling unused ones via kernel command line options by editing the `syslinux.cfg` and `EFI/BOOT/grub.cfg` files on the LAKKA partition? just add `video=<port>:d` after the text `quiet` to disable certain output port (it is enough just to modify the "live" section in each file). to get list of ports you can check first by editing the above files and only adding `textmode retroarch=0` after `quiet` and boot again. after you get the shell prompt just enter this command:

```
for port in /sys/class/drm/*/status ; do echo -n "${port}: " ; cat ${port} ; done
```

it will output something like:

```
/sys/class/drm/card0-HDMI-A-1/status: Disconnected
/sys/class/drm/card0-eDP-1/status: Connected
```

the `<port>` part for the kernel command line option is after `card0-` and before `/`, so in above example the values `HDMI-A-1` and `eDP-1`. It is possible that you have also composite output that is always enabled/connected?

gyllegert, 2022-04-21:
Just to clarify, this is a desktop, but it does have multiple display outputs. HDMI and DP specifically.

I tried getting the status, but none of the folders within sys/class/drm/ contained a status folder. It seems like the device isn't initialized. Since I know the PC is connected using HDMI though I tried writing video=HDMI-A-1 in the cfg files just to test. I still get the Lakka flower with the Lakka (official): <version number> on the screen, but Retroarch never boots up.

After some digging around on various forums it seems like the 5.10 Linux kernel used in Lakka has issues with my specific GPU, and 5.11 doesn't have this issue. So I might just have to wait for an updated version of LibreElec.

ToKe79, 2022-04-21:
please try build from here, it has Kernel version 5.17 - https://nightly.builds.lakka.tv/2022-04-19_LE-master/Generic.x86_64/

gyllegert, 2022-04-21:
I tried but got this error

[image]

I'm not sure what happened, but I tried two different flash drives just to make sure that wasn't the issue.

Thank you for looking into this by the way. :)

ToKe79, 2022-04-21:
this tends to happen when Lakka is installed already on internal drive - or something else interferes with the mount process of tha LAKKA (FAT32) partition. can you try unplugging all internal drives and use only the thumb drive to boot?

gyllegert, 2022-04-21:
Hey!

I tried what you wrote in your edit and it totally worked! Retroarch boots just fine now. Thank you so much for your help, this might be a good temporary workaround for others having the same issue until a new version comes out. Apologies for this becoming more of a support post. It wasn't my intention.

closed 2022-04-21 by gyllegert