libretro/Lakka-LibreELEC #1689: Possible Nightly Update Issue

smilerliu, 2022-04-22:
I have a GPI case + Pi Zero W. I downloaded the 20220422 nightly tar into the .update directory directly from the device using wget and rebooted the device. The update UI showed up and the update seemed to be successful. However, after update, when I view the system information, it says the Lakka version is the 20220422 nightly build, but the first line showing the build time still shows 20220415, and /tmp/cores/race_libretro.so still has a timestamp of 20220330. The 20220422 build should have fixed a bug in the core and I still experience the bug after the update, which makes me feel that the core on my device isn't really updated. I have a feeling that not all the files on my device are updated to the files in the 20220422 nightly build.

I don't even know how to confirm I have a problem or not. Any suggestions?

ToKe79, 2022-04-22:
I checked the `SYSTEM` image file in the tarball:

```
$ cd target/
$ mkdir mnt
$ sudo mount SYSTEM mnt/
$ cd mnt/usr/lib/libretro/
$ stat race_libretro.so
  File: race_libretro.so
  Size: 241488          Blocks: 472        IO Block: 1024   regular file
Device: 700h/1792d      Inode: 3303        Links: 1
Access: (0775/-rwxrwxr-x)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2022-04-22 00:41:22.000000000 +0200
Modify: 2022-04-22 00:41:22.000000000 +0200
Change: 2022-04-22 00:41:22.000000000 +0200
 Birth: -
$ ls -l race_libretro.*
-rw-rw-r-- 1 root root   1074 Apr 13 18:09 race_libretro.info
-rwxrwxr-x 1 root root 241488 Apr 22 00:41 race_libretro.so
```

so the libretro-core file in image is updated, but for some reason the `SYSTEM` file was most probably not updated / copied over to `LAKKA` partition? can you check your `LAKKA` partition what timestamp the `SYSTEM` file has? If it is older, then just download the tarball again, extract it and copy over `target/SYSTEM` to your `LAKKA` partition. also copy the `target/KERNEL` file, but make sure to rename it to the same name as is currently on the `LAKKA` partition (probably `kernel.img`). and also copy all files (except `config.txt`) and folders from `3rdparty/bootloader` folder over to `LAKKA` partition. you can remove the `.md5` checksum files from the root of the `LAKKA` partition, so they don't have false values (as the `SYSTEM` and `kernel.img` have been replaced) and you can (optionally) copy over new `.md5` files from the tarball.

smilerliu, 2022-04-22:
Mystery solved...

1. I mounted the SD card on a computer and verified that SYSTEM in the LAKKA partition is exactly the same in the nightly tarball for 20220422. The size and MD5 both match. So the update process has no problem.

2. I didn't realize overlayfs is used for the cores. Last time when I tested the core you manually built, I moved the old core away, copied the new core to the old core's place, tested, deleted the new core, and moved the old core back. I did all of this to /tmp/cores/. Due to the overlayfs setup, at the end the old core file got copied into ~/cores/ and overrides the core in the SYSTEM file. Once I remove the old core from ~/cores/ and rebooted the device, I can see the new core in /tmp/cores/.

closed 2022-04-22 by smilerliu