libretro/Lakka-LibreELEC #1690: Need mirrors?

karibuTW, 2022-04-23:
Hello,
I'm the admin of mirror.freedif.org and got few servers available in various locations.
Will you need some mirrors to support the projects?

I have 4 locations. Most needs are in Singapore as Asia is usually not that well covered.
But I can do France, Germany or Finland if needed (Usually pretty well covered)

Ping me if needed, I'll be pleased to contribute.

Thank you

Ntemis, 2022-04-23:
Thank you for the kind offer please come to lakka.tv channel in retroarch discord and ping Demetris or Vudiq

karibuTW, 2022-04-27:
Hello, I did twice. No reply so far.
Probably high workload ;)

ToKe79, 2022-05-12:
Hi @karibuTW,

sorry for a late response.

Thank you very much for your offer. I personally do not think that project of this size would benefit from additional mirrors. We have one primary download server for stable releases / direct updates, and other server for dev builds. We also use the GitHub release assets for distribution via RPi Imager.

If other devs think there is additional benefit, please, speak out.

But as the images are freely available, they can be mirrored by anybody and we can just provide links to the mirrors, where the user can then browse for the image they want. I don't think that we will be maintaining list of direct links and redirect the user to a location closest to them.

gouchi, 2022-05-12:
We might use mirror also as an "archive" for old builds.

karibuTW, 2022-05-12:
Thanks @ToKe79 
Ping me if you need any infra. Happy to help

Ntemis, 2022-05-13:
@ToKe79 I wanted this opportunity to remain feasible in case we need to move to a newer server that retroarch team uses which is ~6 times faster of the one we currently have for stable releases until we can enable the new links on the newer server after migration

Ntemis, 2022-06-10:
Thank you for the offer but lets close this

closed 2022-06-10 by Ntemis