libretro/Lakka-LibreELEC #1691: Include PUAE 2021 Core

rsn8887, 2022-04-23:
Can the Commodore - Amige (PUAE 2021) core be included in Lakka distros, especially for Slower devices like ODroid and others? It is useful for games where current PUAE struggles to run at full speed. PUAE 2021 seems to be much faster than the current PUAE, and is already available in Retroarch on the Switch.

gouchi, 2022-04-27:
[Source](https://git.libretro.com/libretro/libretro-uae2021/)

rsn8887, 2022-04-28:
I think here is also the official source code:
https://github.com/libretro/libretro-uae/tree/2.6.1

It is simply branch 2.6.1 of the official PUAE core.

rsn8887, 2022-05-05:
With latest Lakka 4.2, PUAE was updated, and I am now getting slow-down in some Amiga games even on my Pi 400. So inclusion of PUAE 2021 would be a great benefit on all systems, at least Pi 400 and slower.

ToKe79, 2022-05-12:
@rsn8887 the core is now included in test images here: http://nightly.builds.lakka.tv/PRs/pr_1702_000522/

PR #1702 (2022-05-12): puae2021: add new libretro core
puae based on older version for better performance.
closes #1691
closed 2022-05-12 by Ntemis