libretro/Lakka-LibreELEC #1693: Ozone GUI rendering tiny and squeezed on the rg351mp

armadillu, 2022-04-25:
I can't get the `ozone` GUI to look decent using lakka. I am using an rg351mp with the latest release, and I also tested the last nightly build (3.x). Ozone looks squeezed (as if the screen was 16:9, but its 4:3) and way too small. The rg351mp has a 640x480 display.

[image]

I tried ssh'ng in and starting with `retroarch --verbose` to see some logs, and I see retroarch is creating a 640x480 gl context. 
It almost seems like ozone is rendering into a 16:9 fbo and blitting it to the 4:3 screen somehow, it also seems to render too slowly (I get 37fps in ozone). But the logs say otherwise:

```
retroarch --verbose
librga:RGA_GET_VERSION:4.00,4.000000 
ctx=0x86e058,ctx->rgaFd=3 
Rga built version:version:+2017-09-28 10:12:42 
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.10.0 (Git b71be7efd2)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Feb 28 2022
[INFO] Version: 1.10.0
[INFO] Git: b71be7efd2
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 1 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (640x480) 640 x 480, 149.129089 Hz
[INFO] [GL]: Found GL context: "kms".
[INFO] [GL]: Detecting screen resolution: 640x480.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0xf2345f28.
[INFO] [KMS]: New FB: 640x480 (stride: 2560).
[INFO] [GL]: Vendor: ARM, Renderer: Mali-G31.
[INFO] [GL]: Version: OpenGL ES 3.2 v1.r13p0-01rel0.8747d14aec16ec21ac2b2da252c4ec5f.
[INFO] [GL]: Using ARB_sync to reduce latency.
[INFO] [GL]: Using resolution 640x480.
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: "freetype".
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "rk8xx_pwrkey" (/dev/input/event0).
[INFO] [udev]: Keyboard #1: "odroidgo3-keys" (/dev/input/event3).
[INFO] [udev]: Pad #0 (/dev/input/event2) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] ALSA: Using floating point format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
wifi is already enabled
[INFO] [Display]: Found display driver: "gl".
[INFO] [SRAM]: SRAM will not be saved.
[INFO] [Playlist]: Loading history file: "/storage/roms/SD_ROMS/playlists/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/roms/SD_ROMS/playlists/content_favorites.lpl".
[INFO] [KMS]: New FB: 640x480 (stride: 2560).
```

Note that RGUI and XMB render normally.
Trying to increase the Scale Factor only allows me to go up to 1.42, but the text is still too small. Going beyond 1.42 doesnt increase the text size.

I tested EmuElec and ozone renders as you would expect, and renders at 60fps.

`dmesg` shows that screen can run at 640x480@60 and 640x480@149, I am unsure which one is chosen

```
...
[    0.907086] [drm] Rockchip DRM driver version: v1.0.1
[    0.907415] rockchip-drm display-subsystem: devfreq is not set
[    0.907960] rockchip-drm display-subsystem: bound ff460000.vop (ops 0xffffff8008a93b08)
[    0.908034] rockchip-drm display-subsystem: bound ff450000.dsi (ops 0xffffff8008a8b068)
[    0.908046] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    0.908053] [drm] No driver support for vblank timestamp query.
[    0.908290] get_framebuffer_by_node: failed to get logo,offset
[    0.908304] rockchip-drm display-subsystem: can't not find any loader display
[    0.908316] rockchip-drm display-subsystem: failed to show loader logo
[    0.908880] Freeing drm_logo memory: 3648K
[    0.911342] dclk_vopb_frac p_rate(612000000) is low than rate(60000000)*20, use integer or half-div
[    0.919380] rockchip-vop ff460000.vop: [drm:vop_crtc_enable] Update mode to 640x480p149, type: 16
[    0.935577] dclk_vopb_frac p_rate(612000000) is low than rate(60000000)*20, use integer or half-div
[    0.935770] dw-mipi-dsi ff450000.dsi: final DSI-Link bandwidth: 396 x 4 Mbps
[    1.009617] mmc_host mmc1: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
...
[   13.056838] rockchip-vop ff460000.vop: [drm:vop_crtc_enable] Update mode to 640x480p149, type: 16
[   13.057290] dclk_vopb_frac p_rate(23875000) is low than rate(60000000)*20, use integer or half-div
[   13.057483] dw-mipi-dsi ff450000.dsi: final DSI-Link bandwidth: 396 x 4 Mbps

```

I tried forcing a video mode in the kernel boot arguments with `video=DSI-1:640x480@60` but I get the same look in ozone.

blindm0use, 2022-04-28:
#### Sorry for off topic, I do also have the same scaling problem with ozone.

*How exactly did you get Lakka to boot on RG351MP?*

Did you change dtb file with the other one from different os to make it boot?
Do you have volume buttons working?
What about N64 emulation? Can you get it running?

In my case, I had to get dtb from AmberElec and modify extlinux.conf to make my device to boot into Lakka.

armadillu, 2022-04-29:
@blindm0use worked out of the box with the release, just copy image into an sdcard and it works. 
All buttons work, including volume, out of the box. The device vibrates at boot.
Amberelec works too, without the boot vibration.
I tried the dtb swap too (use the one from amberelec in lakka), but as you report, with this dtb the volume buttons didnt work.

blindm0use, 2022-04-29:
Sadly, I am not aware of that, but Kernel Panic can be a result of different hardware revision

blindm0use, 2022-04-29:
Used [this dtb](https://github.com/christianhaitian/arkos/files/8238619/rk3326-rg351mp-linux.zip) and got volume buttons working, but the volume is not balanced with the gain. At max the sound starts cracking

armadillu, 2022-05-03:
To add to this different HW versions discussions, here's a pic of the board inside my rg351mp. It has "RG351MP_v1.3 2021-03-30" printed on it.
[image]

Ntemis, 2023-01-20:
Theme issue take it to ozone developers

closed 2023-01-20 by Ntemis

armadillu, 2023-01-20:
As there's no rg351mp support for the newer versions of lakka, we may as well close this now...

Ntemis, 2023-01-20:
Support is coming with mainline kernel and mesa to OGA family but not everything will be supported as there are a lot of devices and a lot less developers with them

armadillu, 2023-01-21:
Oh that's great to hear! I will check again on the new version (when is testable) and report back