libretro/Lakka-LibreELEC #1698: Lakka not booting on pc

Jonathan20122, 2022-04-29:
I just built a PC, i310100, 16gb ddr4 2666 memory, ASRock h510m/ac motherboard. I could not get LAKKA to boot off of a USB drive. I tested the flashed drive on another computer and it works great. After the screen where you can pick live or installer, nothing shows up. I have tried all display outputs. I also installed to a dedicated ssd using the other machine and that "boots" but just shows a black screen indefinitely as well. Motherboard bios is also up to date, wasn't sure if that could affect it.

vus7294, 2022-04-30:
Is your Driver setting as Vulkan? Lakka on PC Doesn't work on Vulkan :(

Jonathan20122, 2022-04-30:
How do check that or change it.

vus7294, 2022-04-30:
If you Can't boot Lakka, require Mount Linux Partition using Paragon Software's Linux File System for Windows.
And browse Retroarch.cfg and Check Video Setting.

gouchi, 2022-04-30:
You should try to provide some log using [createlog](https://www.lakka.tv/doc/Troubleshooting-Lakka/).

And try to boot with `live textmode ssh` for the boot parameters to make your tests.

Jonathan20122, 2022-04-30:
I determined I can change video settings on the computer that does work and then boot it on the one that doesn't. No video setting fixed the issue. How do I go about creating logs? Also how to I find them after creating so I can paste them here?

Jonathan20122, 2022-05-07:
> You should try to provide some log using [createlog](https://www.lakka.tv/doc/Troubleshooting-Lakka/).
> 
> And try to boot with `live textmode ssh` for the boot parameters to make your tests.

I am unable to boot using that command. Any help would be greatly appreciated.

gouchi, 2022-05-11:
Those  parameters allow you to boot to prompt then you can enter the command [createlog](https://www.lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs).

Jonathan20122, 2022-05-11:
> Those parameters allow you to boot to prompt then you can enter the command [createlog](https://www.lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs).

When I type that in my screen goes totally black, I have attempted typing the second command anyway with no result. Is there any way I can contact you directly? I am happy to compensate you for your help either way.

ToKe79, 2022-05-12:
can you try image from the dev branch?
https://nightly.builds.lakka.tv/2022-04-19_LE-master/Generic.x86_64/
It has much newer kernel.

gouchi, 2022-05-12:
If it not working with the dev branch image, you may are using perhaps Nvidia RTX graphic cards but it won't be supported.

To get best result on PC you have to use AMD or Intel graphic card.

closed 2022-05-16 by Jonathan20122