libretro/Lakka-LibreELEC #1704: Input driver fails to initialize

jclsn, 2022-05-15:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?  Release 4.2

- What system hardware are you using?
  Generic x64

- What did you do?
  Add a per-game remapping

- What did you expect to happen?
  Nothing else

- What happened instead?
  Retroarch can't start and Lakka is stuck at "Waiting for network to come online"

verbose log:

```
[INFO] [Config]: Loading default config.
[INFO] [Config]: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] RetroArch 1.10.3 (Git 9b282aa)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i7-4765T CPU @ 2.00GHz
[INFO] Fähigkeiten:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Apr 25 2022
[INFO] Version: 1.10.3
[INFO] Git: 9b282aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "linuxraw".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version der Libretro-API: 1, Kompiliert gegen API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 6 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: yes
[INFO] [DRM]: Connector 2 has 25 modes.
[INFO] [DRM]: Connector 2 assigned to monitor index: #1.
[INFO] [DRM]: Connector 3 connected: no
[INFO] [DRM]: Connector 3 has 0 modes.
[INFO] [DRM]: Connector 4 connected: no
[INFO] [DRM]: Connector 4 has 0 modes.
[INFO] [DRM]: Connector 5 connected: no
[INFO] [DRM]: Connector 5 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 10: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 12: (1360x768) 1360 x 768, 60.015163 Hz
[INFO] [DRM]: Mode 13: (1280x720) 1280 x 720, 60.000000 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 59.940201 Hz
[INFO] [DRM]: Mode 15: (1280x720) 1280 x 720, 50.000000 Hz
[INFO] [DRM]: Mode 16: (1024x768) 1024 x 768, 60.003841 Hz
[INFO] [DRM]: Mode 17: (800x600) 800 x 600, 60.316540 Hz
[INFO] [DRM]: Mode 18: (720x576) 720 x 576, 50.000000 Hz
[INFO] [DRM]: Mode 19: (720x576i) 720 x 576, 25.000000 Hz
[INFO] [DRM]: Mode 20: (720x480) 720 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 21: (720x480) 720 x 480, 59.940060 Hz
[INFO] [DRM]: Mode 22: (640x480) 640 x 480, 60.000000 Hz
[INFO] [DRM]: Mode 23: (640x480) 640 x 480, 59.940475 Hz
[INFO] [DRM]: Mode 24: (720x400) 720 x 400, 70.081665 Hz
[INFO] [GL]: Found GL context: "kms".
[INFO] [GL]: Detecting screen resolution: 1920x1080.
[INFO] [EGL] Found EGL client version >= 1.5, trying eglGetPlatformDisplay
[INFO] [EGL]: EGL version: 1.5
[INFO] [EGL]: Current context: 0x7f36c8064c00.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: Intel, Renderer: Mesa Intel(R) HD Graphics 4600 (HSW GT2).
[INFO] [GL]: Version: 3.1 Mesa 22.0.2.
[INFO] [GL]: Using resolution 1920x1080.
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ...
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling modern snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling bokeh shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snowflake shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[INFO] [Font]: Using font rendering backend: "freetype".
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[ERROR] [Video]: Cannot initialize input driver. Exiting ...
[ERROR] Fatal error in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 0, Frames dropped: 0.
```

lefetylorantajooleb, 2022-05-26:
I am also having difficulties even with the latest nightly, bluetooth refuses to work and the PS4 controller I've been using not even working with an USB cable.

Ntemis, 2022-06-10:
Tested and working fine here. if you want to open again make sure your ps4 is genuine and dont mess with the driver