libretro/Lakka-LibreELEC #1712: laka freezes than restarts when using the online updater GPI Case Raspberry pi zero W 2

coreybruce, 2022-06-29:
Hey all just wanted to report a ongoing bug on the Gpi Case powered by the Raspberry pi zero W 2.

If you use the online updater to update to the latest nightly build it will freeze than eventually restart the lakka UI and fail to download.

Ntemis, 2022-06-29:
Sure but this cant stay open forever though i"ll give it a few months give or take

coreybruce, 2022-06-29:
Yeah that's understandable, I'm sure the community can help with this

coreybruce, 2022-07-05:
> I've had this problem for a long time, on both Pi Zero W and Pi Zero W 2. The workaround is to just ssh in and use wget to download the update file into the update folder.

Yeah that's pretty much how I've been manually updating but I was hoping the online updater would work which sucks

Ntemis, 2023-01-20:
This is now expired @coreybruce Thank you for the reports

closed 2023-01-20 by Ntemis

coreybruce, 2023-02-04:
Can this please be re-opened, did this get fixed?

ToKe79, 2023-02-04:
Currently there is no plan to fix this, as it is related to the limited memory of the device and how RA downloads the update (caches to memory, when complete dumps to storage). You have to use `lakka-update.sh` script from shell (or wget).

coreybruce, 2023-02-04:
> Currently there is no plan to fix this, as it is related to the limited memory of the device and how RA downloads the update (caches to memory, when complete dumps to storage). You have to use `lakka-update.sh` script from shell (or wget).

Ah so it can't do the updater because it only has 512mbs of ram? I know you said there was no plans to fix this but couldn't this be optimised and just not cache it to memory for low memory devices?