libretro/Lakka-LibreELEC #1722: Mr. Boom on Raspberry Pi 4 on Lakka - Only 5 players supported?

mtoensing, 2022-07-24:
Crosspost from https://github.com/Javanaise/mrboom-libretro/issues/118

- Which version of Lakka are you using? 4.2
- What system hardware are you using? Raspberry Pi 4 with 4 Xbox 360 USB controllers and two usb pads. 
- What did you do? Connected all controllers to the pi and they where all connected with a message in the bottom left. Started MrBoom on Lakka 
- What did you expect to happen? I can play MrBoom with 6 and more players.
- What happened instead? But only 5 players could press "fire" and were recognized. I tried all combinations of controllers. They work all fine in a Bomberman SNES multi-tap game  or a N64 Mario Kart match in any combination. It is not a hardware problem with the controllers.

Ntemis, 2022-07-26:
And why this is a lakka issue? You realize what Lakka is right? Just fyi
Lakka is a lightweight Linux distribution that transforms a small computer into a full blown retrogaming console.
Report this bug to the core github

mtoensing, 2022-08-12:
There was quite the discussion about this. The reason was this:

`input_max_users = "5"`

why is this set to 5 and not 8 in Lakka?

gouchi, 2022-10-15:
Indeed we should [change it to 8](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v5.x/packages/lakka/retroarch_base/retroarch/package.mk#L234) as Mr Boom core supports up to 8 players.

PR #1767 (2022-10-24): retroarch : increase maximum amount of users to 8
Fix #1722