libretro/Lakka-LibreELEC #1723: FTP transfer speed on wired network.

greatliu, 2022-07-29:
Dear author,
I found that if WiFi is connected, the FTP transfer speed on wired network will be limited as same as using WiFi.
It happened on every version of lakka X86.

Hope it will be improved.

gouchi, 2022-08-06:
Hi,

I don't remember we have FTP ? We have SFTP or Samba. 

Which software did you use ? You did try wit Filezilla or Winscp as describe in the [documentation](https://www.lakka.tv/doc/Accessing-Lakka-filesystem/) ?

Can you give us some screenshot for the speed but I am not sure we can optimize it.

Thank you.

greatliu, 2022-08-07:
Hi, frined.
I use Filezilla to transfer.

I do not know how to show you screenshot, sorry. I will describe this problem with detail:
1. connect WiFi;
2. in "network information", lakka will display 2 IP: one is wlan, another is eth0;
3. transfer files by Filezilla using eth0's IP, you will find the speed is the same as using WiFi;
4. get off wifi, then reboot lakka, transfer files by Filezilla using eth0's IP, the speed is recovered to gigabyte.

I just tried it, the problem is still as I said.

Thanks.

barrystaes, 2022-12-24:
@greatliu what happens if you connect to a unrelated WiFi network such as your phone hotspot, and then try to FTP with something on your LAN? In case you have a VPN running on Lakka disable it temporarily.

closed 2023-01-20 by Ntemis