libretro/Lakka-LibreELEC #1724: chinese characters display bug.

greatliu, 2022-08-01:
Dear author,
I found that when I use Chinese UI language, some characters displayed as "口". Especially in the pop-up message in the left-buttom.
Can you improve this bug?

all the X86 lakka version have this bug, since lakka supported Chinese.

gouchi, 2022-08-06:
Hi,

Can you try [this](https://github.com/libretro/Lakka-LibreELEC/issues/1354#issuecomment-880622077) if you are using xmb menu ?

You can find some fonts in `/usr/share/retroarch-assets/pkg`

Thank you.

greatliu, 2022-08-07:
Hi, firend.
As your reply, I just have tried all the fonts in /usr/share/retroarch-assets/pkg. Unfortunately, all of them diaplay Chinese characters as "口".
Can you help?

gouchi, 2022-08-07:
If you are using xmb you need to restart RetroArch after you changed the font.

greatliu, 2022-08-07:
> If you are using xmb you need to restart RetroArch after you changed the font.
I confirm restarted, the problem is still unsolved.

gouchi, 2022-08-07:
I can't reproduce your issue using [nightly build](https://nightly.builds.lakka.tv/2022-08-04/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20220804-9e969c4.img.gz) and `chinese-fallback-font.ttf`.

[image]

You have to restart RA

[image]

And Ozone menu is working also in Chinese.

[image]

greatliu, 2022-08-07:
> I can't reproduce your issue using [nightly build](https://nightly.builds.lakka.tv/2022-08-04/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20220804-9e969c4.img.gz) and `chinese-fallback-font.ttf`.
> 
> [image]
> 
> You have to restart RA
> 
> [image]
> 
> And Ozone menu is working also in Chinese.
> 
> [image]

My version is 4.2 release, there were no chinese-fallback-font.ttf in dir. I downloaded some kinds of Chinese fonts and tested (sorry I did not find chinese-fallback-font.ttf to download). It runs, the main UI displays correctly, but the pop-up message in the left-buttom, there were some "口" still.
e.g. when I select reboot, the pop-up displays as "正在重口", it run the same as the default fonts.
I always use release ver, and did not test nightly ver. Next if I have idle time, I will test nightly ver and chinese-fallback-font.ttf.

Thank you very much for your help and patience,

closed 2023-01-20 by Ntemis