libretro/Lakka-LibreELEC #1725: Cannot control volume by keyboard hotkey.

greatliu, 2022-08-02:
Dear author,
I found Lakka cannot recognize the "volume+" and "volume-" button, so when I run Lakka on some laptop or all-in-one computer, I need to use Lakka menu to control volume.
I tried, and found that Lakka recognize the "volume+" and "volume-" hotkeys on all keyboards.
I think it is a convenient function for laptop or all-in-one computer users.

Can you add this function? Thanks.

gouchi, 2022-08-06:
Hi,

By default, it should be `-` and `+` . You can change it from `Settings tab > Input > Hotkeys > Volume Up/Down`.

Thank you.

greatliu, 2022-08-07:
> Hi,
> 
> By default, it should be `-` and `+` . You can change it from `Settings tab > Input > Hotkeys > Volume Up/Down`.
> 
> Thank you.

Hi,
Thanks for your notify.

But if I use lakka on an all-in-one computer without keyboard, it is still a problem.
Can lakka support  "volume+" and "volume-" keyboard hotkeys?

gouchi, 2022-09-19:
I will close this issue as there is [one](https://github.com/libretro/RetroArch/issues/14298) opened on RetroArch issue tracker.

closed 2022-09-19 by gouchi