libretro/Lakka-LibreELEC #1726: Segfault when switching to any of the OpenGL drivers on AMD GPU

GPDP1, 2022-08-05:
I recently acquired a Radeon RX550 GPU and promptly installed it on my generic x86 desktop. About half a year ago I had installed an R7 250, which was working well, but the fan died on me, so between then and now I've been using the integrated Intel HD 630 graphics, again with no issues. Upon first installation, all seemed well, as it was already set to use the Vulkan driver, which gives me no problems on any of the cores and I can even use the ParaLLEl-RDP renderer on Mupen64Plus-Next. However, when I attempted to try out GLideN64 (which automatically makes RetroArch switch to the glcore driver), RetroArch crashed, and on further testing I then noticed it always crashes even on something like SNES if I attempt to switch to any of the three GL drivers (gl1, gl and glcore). I tried starting RetroArch with RADV_DEBUG=llvm to see if that worked better, but to no avail. Once again, none of this happens when using Vulkan.

Here's the relevant output from the log:

> [INFO] [Video]: Set video size to: 3840x2160.
> [INFO] [DRM]: Found 4 connectors.
> [INFO] [DRM]: Connector 0 connected: no
> [INFO] [DRM]: Connector 0 has 0 modes.
> [INFO] [DRM]: Connector 1 connected: no
> [INFO] [DRM]: Connector 1 has 0 modes.
> [INFO] [DRM]: Connector 2 connected: no
> [INFO] [DRM]: Connector 2 has 0 modes.
> [INFO] [DRM]: Connector 3 connected: no
> [INFO] [DRM]: Connector 3 has 0 modes.
> [WARN] [DRM]: Couldn't get device connector.
> [WARN] [DRM]: Couldn't get device resources.
> [INFO] [DRM]: Found 3 connectors.
> [INFO] [DRM]: Connector 0 connected: no
> [INFO] [DRM]: Connector 0 has 0 modes.
> [INFO] [DRM]: Connector 1 connected: no
> [INFO] [DRM]: Connector 1 has 0 modes.
> [INFO] [DRM]: Connector 2 connected: yes
> [INFO] [DRM]: Connector 2 has 46 modes.
> [INFO] [DRM]: Connector 2 assigned to monitor index: #1.
> [WARN] [DRM]: Couldn't find DRM encoder.
> [WARN] [DRM]: Couldn't get device resources.
> [ERROR] [KMS]: Couldn't find a suitable DRM device.
> [INFO] [GLCore]: Found GL context: "null".
> [INFO] [GLCore]: Detecting screen resolution: 320x240.
> [INFO] [GLCore]: Vendor: (null), Renderer: (null).
> [INFO] [GLCore]: Version: (null).

And here are the relevant logs:
[log-2022-08-05-00.28.17.zip](https://github.com/libretro/Lakka-LibreELEC/files/9264355/log-2022-08-05-00.28.17.zip)

Let me know if you need me to try anything else.

gouchi, 2022-08-06:
Hi,

Do you reproduce this issue with this [build](https://nightly.builds.lakka.tv/members/gouchi/master/Lakka-Generic.x86_64-LE-master-devel-20220724114930-4f0cf20.img.gz) from development branch ?

Thank you.

GPDP1, 2022-08-07:
> Hi,
> 
> Do you reproduce this issue with this [build](https://nightly.builds.lakka.tv/members/gouchi/master/Lakka-Generic.x86_64-LE-master-devel-20220724114930-4f0cf20.img.gz) from master branch ?
> 
> Thank you.

I just tried it on a live USB, and indeed, the OpenGL drivers work just fine with that build. They also don't seem to suffer from an issue I just discovered with the Vulkan driver where the video output to the screen seems to go out if I so much as touch the audio latency settings for some reason, but that's another problem I'll be opening another ticket for soon.

GPDP1, 2022-08-07:
Update: since the live LE-master image seemed to work, I decided, just to cross more things off the list, to try booting a fresh nightly 4.x image, and, what do you know, GL drivers work there as well! That's when it hit me: these live booting images are set to the gl driver by default, whereas when I first installed my card, I started with Vulkan and then tried switching to gl or glcore from within the UI. So I went back to my regular installation, modified retroarch.cfg so it would start with glcore instead of Vulkan, and RetroArch started just fine. I was able to start games, use GLideN64, shaders, everything. I then tried switching to Vulkan, and that also worked. Then I switched from Vulkan back to glcore, and yup, it crashes, same as before.

So yeah, the problem seems to be with the act of switching FROM Vulkan TO any of the GL drivers. Starting RetroArch with gl or glcore gives me no issues, and switching from the gl drivers to Vulkan is also ok, but Vulkan to gl/glcore is the problem.

gouchi, 2022-08-07:
Hi,

Thank you for the feedback.

You may try to get a diff of [retroarch.cfg](https://www.lakka.tv/doc/Configuring-Lakka/) to "pinpoint" the issue ?

gouchi, 2022-08-07:
Anyway you should be better use `glcore` instead of `gl` driver, if you have to use opengl driver.

GPDP1, 2022-08-07:
> Anyway you should be better use `glcore` instead of `gl` driver, if you have to use opengl driver.

Oh yeah, I'm aware. Only reason I'm even talking about gl is to drive home the point that this problem happens when switching to any of the OpenGL drivers and not just glcore. I didn't buy a modern-ish GPU just to limit myself to OpenGL 2.x support lol

> You may try to get a diff of [retroarch.cfg](https://www.lakka.tv/doc/Configuring-Lakka/) to "pinpoint" the issue ?

Sorry, but by this, do you mean take both a working retroarch.cfg and a non-working one, and point out the difference between them?

gouchi, 2022-08-07:
> Sorry, but by this, do you mean take both a working retroarch.cfg and a non-working one, and point out the difference between them?

Correct ;)

GPDP1, 2022-08-08:
Save for this one thing, everything else in retroarch.cfg is exactly the same:

> video_driver = "glcore"

RetroArch starts, all works well.

> video_driver = "vulkan"

RetroArch also starts and works, but if I try to switch to `glcore` within the UI, RetroArch crashes.

I forgot to mention, when it crashes, it restarts and goes back to `vulkan`.

gouchi, 2022-08-15:
Thank you, we will need more investigation to solve this issue because there are not a lot of changes in the configuration file. 

We should try to make a test when a new stable version is released to see if you reproduce the issue.

GPDP1, 2022-12-24:
I just tried this again on the latest nightly, and the issue appears to be gone at last. I can switch between both glcore and vulkan drivers without any problems, even on hardware-accelerated cores like Mupen64Plus-Next. I think this issue can be closed.

gouchi, 2022-12-24:
Thank you for the feedback.

closed 2022-12-24 by gouchi