libretro/Lakka-LibreELEC #1727: Please add encryption/luks support for file systems

pasnox, 2022-08-05:
LibreElec, on which is based Lakka does not support encryption / luks by default.
Though, there is a small community fork that enable that.
I would like to see lakka supporting encrytion as well.
See https://forum.libreelec.tv/thread/20338-le-9-2-with-lvm2-luks-dm-crypt-veracrypt-ext4-encryption-and-some-newclock5-kodi/ for more information.

Ntemis, 2022-08-05:
Am not against this but i think you should take this upsteam https://github.com/LibreELEC/LibreELEC.tv/issues
Then we can have it and stay maintained so a win-win  for everyone.
Also feel free to send them a pr if you like.

pasnox, 2022-08-06:
They want not that feature in LE - not enough user friendly 😭

closed 2024-04-04 by ToKe79