libretro/Lakka-LibreELEC #1728: Lakka 4.2 Gamepad Input Issue

Dootdootshaboot, 2022-08-07:
Inputs were working just fine but after I tried to use a different controller, Switch Pro Controller, everytime I pressure the A button on the controller to select whatever in the game, it pauses the whole game via the Lakka OS controls. 
Somehow it's binded the Pause Button to the most crucial button needed to play any game with a controller: the A Button. Now anytime I press it, I can't play the game bc it'll pause it. To press it, I have to press the hotkey on the keyboard to unpause it. This isn't really usable like this. 
I've already tried to rebind every key on the gamepad/ change the Pause key, but it still did not work. 
I think it's a bug? But I seriously can't play games now

gouchi, 2022-08-07:
Can you try if you reproduce the issue with this [testing build](https://nightly.builds.lakka.tv/members/gouchi/master/Lakka-Generic.x86_64-LE-master-devel-20220724114930-4f0cf20.img.gz) ?

Also did you try with another controller driver in Settings tab > Drivers > Controller ?

Dootdootshaboot, 2022-08-07:
> Can you try if you reproduce the issue with this [testing build](https://nightly.builds.lakka.tv/members/gouchi/master/Lakka-Generic.x86_64-LE-master-devel-20220724114930-4f0cf20.img.gz) ?
> 
> Also did you try with another controller driver in Settings tab > Drivers > Controller ?

I can try. I havent messed wih the drivers bc i couldnt figure out how to delete any of the controler input saves or something. 
Lakka is downlaoded to the harddrive of the computer now, so i can try  and run the test build off of a USB

auxfam, 2022-09-05:
incidently: I raised similar button duplication trigger under https://github.com/libretro/RetroArch/issues/14373  which might be relevant (I only use lakka so maybe it;s actually on other OS as well)  this one though takes SELECT -> L2 and START->R2

Ntemis, 2023-01-20:
User fault , no logs

closed 2023-01-20 by Ntemis