libretro/Lakka-LibreELEC #1731: Error while compiling

turtle72, 2022-08-19:
I was compiling the newest version of lakka(atm) and i got this error
FAILURE: scripts/build same_cdi:target during make_target (package.mk)
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_TARGET}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install same_cdi:target has failed!

[312.log](https://github.com/libretro/Lakka-LibreELEC/files/9378480/312.log)

Thoughts?

gouchi, 2022-08-21:
Hi,

Can you retry with

`DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 scripts/clean same_cdi`

then

`DISTRO=Lakka PROJECT=RPi DEVICE=RPi4 ARCH=aarch64 scripts/build same_cdi`

Thank you.

Ntemis, 2022-10-20:
No feedback since open, closing down

closed 2022-10-20 by Ntemis