libretro/Lakka-LibreELEC #1732: 4.2 fresh installation won't boot

danielemegna, 2022-08-21:
> Which version of Lakka are you using?

Latest official stable build Lakka-RPi.arm-4.2

> What system hardware are you using?

Raspberry Pi 1 Model B

> What did you do?

Flash the downloaded image `Lakka-RPi.arm-4.2.img.gz` via Balena Etcher on a large sd card (64gb) and boot it.

> What did you expect to happen?

First boot with resizing file system phase, automatic reboot and first boot on retroarch interface.

> What happened instead?

- succefully first boot with resizing file system phase
- automatic reboot
- failure on any now os boot with just this log on screen

```
Lakka (official): 4.2
[  TIME  ] Timed out waiting for device /dev/serial1.
[DEPEND] Dependency failed for Broadcom sdio firmware update for BCH43430A1.
[FAILED] Failed to start Samba NMB Daemon.
[FAILED] Failed to start Samba MMD Daemon.
[FAILED] Failed to start Samba SMB Daemon.
[FAILED] Failed to start Samba NMB Daemon.
[FAILED] Failed to start Samba NMB Daemon.
[FAILED] Falled to start Samba SMD Daemon. [FAILED] Failed to start Sanba NMB Dacnon.
```

[image]

Am I doing something wrong?

danielemegna, 2022-08-21:
Just given a try also to latest nightly build `Lakka-RPi.arm-nightly-v4.x-20220804-9e969c4`, same behavior.

gouchi, 2022-08-29:
Hi,

I could make a test on RPi  1 Model B+ with `Lakka-RPi.arm-4.2.img.gz` and it is working.

You should try to ask some feedback on our [Discord channel](https://discord.gg/BNFR4hM) if possible.

Thank you.

Kind regards,

danielemegna, 2022-09-02:
Ouch.. the only different seems to be `Model B` vs `Model B+` :disappointed: 
Thank you @gouchi anyway.

nayaabkhan, 2022-10-28:
@danielemegna I'm having the same issue, I see you closed the issue. Could you please guide me on how did you manage to solve this?

danielemegna, 2022-10-28:
No, I didn't solve sadly .. let me know if you discover something new

nayaabkhan, 2022-10-28:
Ah, that's a bummer. I'll be sure to update if I find a fix. Meanwhile, I was wondering if it is possible for you to reopen this issue. If not I'll file a new one.

nayaabkhan, 2022-10-31:
So I had no luck fixing the issue on first boot (after partition resizing). It is stuck on "Failed to start Samba SMB Daemon." forever. So I went for compiling RetroArch instead. After a few days of work, I've successfully compiled it and it works like a charm!

For anyone else looking to put their Raspberry Pi 1 to work and has issues with Lakka, they can follow the steps here:
https://github.com/nayaabkhan/rpi1b-retroarch

Ntemis, 2023-01-20:
can you guys try the 4.3 release

danielemegna, 2023-02-02:
I'm giving a try to latest `Lakka-RPi.arm-4.3.img` for _Raspberry Pi (arm)_ . I'll update you soon.

danielemegna, 2023-02-02:
Sadly nothing changed.

Nice and smooth first boot with resize and reboot phases:
[image]

Stuck on every next boot, also waiting for several minutes:

[image]

:sob: :sob: :sob:

danielemegna, 2023-02-02:
> For anyone else looking to put their Raspberry Pi 1 to work and has issues with Lakka, they can follow the steps here:
> https://github.com/nayaabkhan/rpi1b-retroarch

could you please restore the content on this link @nayaabkhan ? https://github.com/nayaabkhan/rpi1b-retroarch

nayaabkhan, 2023-02-03:
@danielemegna Uh oh, the repo was private. Now should be OK.

gouchi, 2023-12-19:
Hi,

Is it working with [Lakka 5.x build](https://nightly.builds.lakka.tv/devbuild_latest/RPi.arm/) ?

Thank you.

danielemegna, 2023-12-20:
> Hi,
> 
> Is it working with [Lakka 5.x build](https://nightly.builds.lakka.tv/devbuild_latest/RPi.arm/) ?
> 
> Thank you.

Ok @gouchi  I'll give a try in the weekend and I'll update you :+1:

danielemegna, 2023-12-31:
Sadly is going worst. Immediate kernel panic at the first boot for missing shared library `libzstd.so`:

[image]

ToKe79, 2024-04-17:
this might be related to neon support - probably model b does not support neon. anyway, nothing that we can/will fix most probably.

closed 2024-04-17 by ToKe79