libretro/Lakka-LibreELEC #1734: Addiing Unico Fun R1 Support

iokzero, 2022-08-30:
Can we add Support Unico Fun R1 Support? 

Works on supporting the device is basically done under other elec-base OS. (except AV port seens still not working)
Original Request from Batorcera (with main contributor's blessing to have it port to other OS)

https://github.com/batocera-linux/batocera.linux/issues/6819#issuecomment-1211670408

Codes:
https://github.com/batocera-linux/batocera.linux/blob/master/board/batocera/amlogic/linux_patches/0119-dt-bindings-arm-amlogic-add-FUN-R1-bindings.patch
https://github.com/batocera-linux/batocera.linux/tree/master/board/batocera/amlogic/s905/fun-r1

Other useful link:
https://github.com/batocera-linux/batocera.linux/blob/master/package/batocera/core/batocera-system/Config.in

It would be great if Lakka can support this interesting SNK official licenced Console. (Unico are the same folks who make the neogeo mini, Arcade strick Pro, and MVSX)

Ntemis, 2022-09-26:
This can be now achieved but we need a dedicated tester for this target

iokzero, 2022-09-26:
> This can be now achieved but we need a dedicated tester for this target

Depended on what you needed for testing, I have the actual hardware, we can run build on