libretro/Lakka-LibreELEC #1735: WiFi Driver won't work on Banana PI M2 Zero

spn0381, 2022-09-01:
Hi, I'm trying to install Lakka on my Banana PI M2 Zero, (the stable, and also the last nightly) and the embedded wifi don't find any network.

- Which version of Lakka are you using? the problem is on the stable version (Lakka-H2-plus.arm-4.2-bananapi-m2-zero) and the last nightly (Lakka-H2-plus.arm-nightly-v4.x-20220804-9e969c4-bananapi-m2-zero)
- What system hardware are you using? Banana PI M2 Zero

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

I have tryed to switch driver but nothing happens, any suggestion?

I cannot collect logs because I haven't the possibility to connect to wifi :( :(

by-gnome, 2022-11-13:
PR LibreELEC/brcmfmac_sdio-firmware/pull/28 will add required files files to brcmfmac_sdio-firmware package.
After merging we will update this package in our repo.

by-gnome, 2022-11-13:
Need to find a way to get the `dmesg` log.

by-gnome, 2022-11-17:
The kernel config seems to be ok:
```
CONFIG_CFG80211=m
CONFIG_MAC80211=m
CONFIG_BRCMUTIL=m
CONFIG_BRCMFMAC=m
CONFIG_BRCMFMAC_SDIO=y
```
Not sure about `CONFIG_RFKILL` which is not set, but I guess it is not needed.

There is `brcmf` node in device tree. So adding nvram and updating brcmfmac_sdio-firmware should fix the problem.

@spn0381 Please try the latest nightly build.

closed 2023-01-20 by Ntemis