libretro/Lakka-LibreELEC #1736: Bluetooth Driver won't work on Nano PI K2

inthesun29, 2022-09-07:
> Lakka:~ # dmesg | grep "brcm"
[    7.832723] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    7.850192] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    7.893989] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.friendlyarm,nanopi-k2.txt failed with error -2
[    8.042236] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    8.042436] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    8.046756] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct  8 2016 15:31:47 version 7.46.57.4.ap.r4 (A1 Station/P2P) FWID 01-3621395e es6.c5.n4.a3

I can't found the brcmfmac43430-sdio.friendlyarm,nanopi-k2.txt file in brcm directory.

by-gnome, 2022-11-12:
PR LibreELEC/brcmfmac_sdio-firmware#26 will add required files files to brcmfmac_sdio-firmware package.
After merging we will update this package in our repo.

by-gnome, 2022-11-12:
PR https://github.com/LibreELEC/brcmfmac_sdio-firmware/pull/28 will add required files files to brcmfmac_sdio-firmware package.
After merging we will update this package in our repo.

by-gnome, 2022-11-13:
Adding brcmfmac43430-sdio.friendlyarm,nanopi-k2.txt wil not solve bluetooth problem.
Kernel should load patchram for BCM43430, someway like this:
```
[   18.327769] Bluetooth: HCI UART driver ver 2.3
[   18.327799] Bluetooth: HCI UART protocol H4 registered
[   18.328060] Bluetooth: HCI UART protocol LL registered
[   18.328228] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   18.328690] Bluetooth: HCI UART protocol Intel registered
[   18.333354] Bluetooth: HCI UART protocol Broadcom registered
[   18.333699] Bluetooth: HCI UART protocol QCA registered
[   18.333729] Bluetooth: HCI UART protocol AG6XX registered
[   18.334051] Bluetooth: HCI UART protocol Marvell registered
[   18.813643] Bluetooth: hci0: BCM: chip id 62
[   18.821358] Bluetooth: hci0: BCM: features 0x0f
[   18.845643] Bluetooth: hci0: BCM4330B1
[   18.845690] Bluetooth: hci0: BCM4330B1 (002.001.003) build 0000
[   18.943286] Bluetooth: hci0: BCM4330B1 'brcm/BCM4330B1.hcd' Patch
[   25.579569] Bluetooth: hci0: Win8 Tablet Semco SWB-B42G - BCM4330B1 37.4 MHz Class 2 NoExtLNA
[   25.579605] Bluetooth: hci0: BCM4330B1 (002.001.003) build 1316
```
We need to check the kernel config and device tree.

by-gnome, 2022-11-16:
The kernel config is ok:
```
CONFIG_BT_BCM=m
CONFIG_BT_HCIUART=m
CONFIG_BT_HCIUART_BCM=y
```
But `uart_A` is disabled in device tree.

by-gnome, 2022-11-19:
@inthesun29 Please try the latest nightly build.

Ntemis, 2023-01-20:
I consider the issue fixed by now, closing down

closed 2023-01-20 by Ntemis