libretro/Lakka-LibreELEC #1738: My new XFX Radeon RX 6400 Video Card hangs at flower

TinkerStuff, 2022-09-10:
Full Video Card Product Finder:
XFX SPEEDSTER SWFT105 Radeon RX 6400 4GB GDDR6 PCI Express 4.0 Low Profile Video Card RX-64XL4SFG2

Is this card currently supported?  
I verified it works booting to Windows10 just fine, proper drivers installed.
My previous RX 4100 Pro worked fine.  
The RX 6400 seems to be a popular card for mini gaming rigs now since its one of the very few modern single-slot low-profile cards available. 

I tried the most recent devel build hoping for a fix contained within, but sadly no.
devel-20220908123921-9e969c4

I've tried stopping the service, editing the retroarch.cfg file to change to various video_driver="gl" (and "glcore", "vulkan") but still stuck.

I have attached the relevant logs.

[log-2022-09-10-08.39.55.zip](https://github.com/libretro/Lakka-LibreELEC/files/9539912/log-2022-09-10-08.39.55.zip)

Thanks in advance!

gouchi, 2022-09-10:
Hi,

May you try to make a test with this [testing build](https://nightly.builds.lakka.tv/members/gouchi/master/Lakka-Generic.x86_64-LE-master-devel-20220724114930-4f0cf20.img.gz) ?

Thank you.

TinkerStuff, 2022-09-10:
Thanks for such a quick response!  
The menu loads with gl / glcore drivers.  Games seems to work fine.
Vulkan driver won't load the menu.

Again, Thank you!

TinkerStuff, 2022-09-10:
Leaving open until the fix appears in one of the nightly releases, or unless OK to close now.

Ntemis, 2023-01-20:
Should be fixed once 5.x nightlies start rolling

gouchi, 2023-01-25:
Can you try to make a test with [dev build from Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) ?

Thank you.

Axallon, 2023-02-09:
Hello, I also recently got an XFX Speedster Radeon RX 6400 Low Profile, hoping for better Vulkan compatibility. I used your link to the Lakka 5.x devbuild. It works! Lakka loads to menu, I can switch the video driver to Vulkan, reboot, and when I go back into the video drivers it detects my new AMD Radeon RX 6400.
I did notice the LAKKA_DISK partition is only 34 MB, so I cannot copy any of my PS2 ROMs to it. My Linux PC will not let me repartition it.
Is there another 5.x version of the devbuild that will allow more space on the LAKKA_DISK partition? I would like to test Vulkan compatibility with LRPS2 (PCSX2).

ToKe79, 2023-02-10:
LAKKA_DISK should be expanded to the full size of your USB thumb drive during the first boot. If it did not, flash the image again and boot from it. you should see messages on screen during the first boot regarding the expansion of the second partition followed by a countdown and reboot.

Axallon, 2023-02-10:
I have used Lakka v4.2 and v4.3 and it does just that - on the first boot it creates the LAKKA_DISK partition and automatically expands to fill the remaining disk space. The 5.x devbuild, however, did not. It created LAKKA_DISK, but with only 34 MB partition.
[image]

ToKe79, 2023-02-11:
@Axallon can you check the ext4 partition on a Linux machine? There should a zero-sized hidden (dot) file - `.please_resize_me`. If it is not there, create it. This tells the init script that the partition needs resizing. Also there should be **no** `.kodi`, `.config` and `.cache` folder - if these exist, **remove them**. When these conditions are met, during boot the partition should be resized to the remaining space of the storage device.

Axallon, 2023-02-13:
What file extension (filetype) does the .please_resize_me file need to be? I tried a folder and a txt, but neither worked.

ToKe79, 2023-02-13:
It must be a file (not a folder) without any extension.

gouchi, 2023-09-20:
@TinkerStuff @Axallon If it is ok with you we can close this issue as it is fixed in Lakka 5.x build ?

gouchi, 2023-12-19:
Closing this issue as it is working using Lakka 5.x.

closed 2023-12-19 by gouchi