libretro/Lakka-LibreELEC #1750: External Video Output with Monitor Index Auto config

arthurlozano, 2022-10-01:
When pluging HDMI, retroarch interface doesn't switch to new monitor automatically.

Then we have to go to "Configuration -> Video -> Output -> Monitor Index -> 2" to display on the HDMI
The problem is when we disconnect this video, lakka keeps trying to output to index 2, even if there's no display connected.

So, what Lakka should do was change the interface of retroarch to the new plugged monitor ou better yet, have some sort of order to respect when monitor index is set to 0 (Auto) to output video, like HDMI, DP and VGA FIRST of LVDS for example.

I tried to play with this command, and change xrandr primary display, but xrandr is not accesible through SSH: 
for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done

Just for clarification, this command outputs this when no external display is connected:
DP-1: disconnected
HDMI-A-1: disconnected
LVDS-1: connected
VGA-1: disconnected

and when I plug my TV via HDMI it shows:
DP-1: disconnected
HDMI-A-1: connected
LVDS-1: connected
VGA-1: disconnected

But nothing happens when video_monitor_index is set to 0, I still have to change manually, causing problems when I disconnect from HDMI.

I hope I could be clear with this issue, sorry for my bad english, I'm still learning :)

gouchi, 2022-11-12:
Hi,

There are two related issues [1](https://github.com/libretro/RetroArch/issues/6271) [2](https://github.com/libretro/RetroArch/issues/12635) on RetroArch about monitor index issue but we are using DRM/KMS not Xorg or Wayland.

We don't have something to switch automatically to the new monitor.

And I am not sure you can achieve it using for example drm/kms tool [modetest](https://wiki.st.com/stm32mpu/wiki/DRM_KMS_overview#modetest_-28DRM-2FKMS_test_tool-29).

gouchi, 2023-02-14:
You might try to make some test with [Lakka 5.x dev build](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/) with x11 support.

Thank you.

ToKe79, 2023-02-14:
Only thing I can suggest in this case (as there is no automatic display switching implemented), before unplugging HDMI from TV, change the monitor index in RetroArch to the ID of the internal panel, probably 1.

ToKe79, 2024-04-04:
no response

closed 2024-04-04 by ToKe79