libretro/Lakka-LibreELEC #1765: Lakka for Odroid Go Ultra

Relwi, 2022-10-18:
The recent gaming platform by HardKernel, Odroid Go Ultra https://www.hardkernel.com/blog-2/odroid-go-ultra/

Ntemis, 2022-10-20:
We have requested Odroid to send over some samples but they haven't respond back yet.

Ntemis, 2022-10-27:
@Relwi https://forum.odroid.com/viewtopic.php?f=221&t=45540
Feel free to comment there if you want to show your interest.

Relwi, 2022-10-28:
Great, thanks!

NeoTheFox, 2023-04-24:
Is there someone working on it already? If there's a branch I would like to know/contribute

Relwi, 2023-04-25:
@NeoTheFox I think they are on it. I recommend to join their discord server for more info

raidenii, 2024-06-10:
Bump - is this dead?

NeoTheFox, 2024-06-10:
> Bump - is this dead?

You should look at ROCKNIX https://github.com/ROCKNIX/distribution

raidenii, 2024-06-10:
>You should look at ROCKNIX
Well, the reason I'm asking is that Lakka's UI is entirely Retroarch - I do not want EmuStation at all.

Ntemis, 2024-06-10:
The issue with this one was that initially the mainline uboot for this device had a serious bug and the boards for porting lakka into it were burned out and no longer usable.
Same issue happened on other ditros too https://forum.odroid.com/viewtopic.php?f=223&t=48481
Dont give up hope yet ;)

GavinDarkglider, 2024-06-28:
> The issue with this one was that initially the mainline uboot for this device had a serious bug and the boards for porting lakka into it were burned out and no longer usable.
> Same issue happened on other distros too https://forum.odroid.com/viewtopic.php?f=223&t=48481
> Dont give up hope yet ;)

Mine isn’t dead, but I put it somewhere for a later date, because of the issues above, and have no idea where I put it. I have a feeling when I find it, it will be with the Odin pro with the v3 screen….. And that is assuming my ex roommate didn’t run off which them.