libretro/Lakka-LibreELEC #1766: please add x11support

greatliu, 2022-10-23:
Hi, dear author.
Can you add x11 support for x86 version?
I got a brandnew CRT and want to use the switchRES function. But I can not find a lakka build with X11 driver.

Thanks.

Ntemis, 2022-11-11:
you need to build it your own, can you do that?

greatliu, 2022-11-24:
> Hi, sir.
I have tried compile lakka, but there is a "keyerror".
Can you tell me what I should do?
Thanks.

$ DISTRO=Lakka PROJECT=Generic ARCH=x86_64 NVIDIAX11=yes make image
./scripts/image mkimage

 =================================================================================
 Configuration for Lakka (community)
 =================================================================================

 Buildsystem configuration:
 ======================================================
 - CPU:                                  x86-64
 - Kernel Architecture:                  x86
 - Userland Architecture:                x86_64
 - CPU features:                         64bit cmov cx8 fpu fxsr mmx no_sahf osfxsr sce sse sse2
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:         yes
 - MOLD (Modern Linker) Support:
 - Default Linker:
 - LLVM support:                         yes
 - DEBUG:                                no
 - CFLAGS:                               -march=x86-64 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG
 - LDFLAGS:                               -Wl,--as-needed -fuse-ld=bfd

 Misc. hardware configuration:
 ======================================================
 - ALSA support:                         yes
 - Pulseaudio support:                   no
 - Bluetooth support:                    yes
 - Include driver:                       RTL8192DU
 - Include driver:                       RTL8812AU
 - Include driver:                       bcm_sta
 - Include firmware:                     misc-firmware
 - Include firmware:                     wlan-firmware
 - Include firmware:                     dvb-firmware
 - Include firmware:                     iwlwifi-firmware

 Misc. Filesystems:
 ======================================================
 - Swap Support:                         yes
   - Swapfile default size:              128
 - Install HFS Tools:                    yes

 Network service configuration:
 ======================================================
 - Avahi (Zeroconf) support:             yes
 - NFS mounting support:                 yes
 - SAMBA mounting support:               yes
 - SAMBA server support:                 yes
 - SFTP server support:                  yes
 - OpenVPN support:                      yes
 - WireGuard support:                    yes

 Graphic configuration:
 ======================================================
 - Graphic Drivers:                      crocus i915 iris r300 r600 radeonsi virtio vmware
 - Display Server:                       no
 - Window Manager / Compositor:          no
 - OpenGL (GLX) support (provider):      no
 - OpenGL ES support (provider):         yes (mesa)
 - Vulkan API support (provider):        yes (vulkan-loader)
 - Vulkan Graphic Drivers:               amd intel

 Video Acceleration configuration:
 ======================================================
 - VA-API Support:                       yes
 - VDPAU Support:                        no

 OS configuration:
 ======================================================
 - OEM Support:                          no
 - Default ROOT Password:                root
 - Bootloader:                           syslinux
 - UDevil support:                       yes
 - Installer support:                    yes
 - Include package:                      rsync
 - Include package:                      xpadneo

 Lakka configuration:
 ======================================================
  Including optional packages:
                                retroarch_assets
                                retroarch_overlays
                                retroarch_joypad_autoconfig
                                libretro_database
                                core_info
 ======================================================
  Including 128 libretro core(s)

 =================================================================================
 End Configuration for Lakka
 =================================================================================

Packages loaded : 1083
Build trigger(s): 1 [image]
Package steps   : 388

Traceback (most recent call last):
  File "scripts/pkgbuilder.py", line 918, in <module>
    result = builder.build()
  File "scripts/pkgbuilder.py", line 453, in build
    job = self.getCompletedJob()
  File "scripts/pkgbuilder.py", line 559, in getCompletedJob
    job = self.complete.get(block=True, timeout=self.captureStats(finished=False))
  File "scripts/pkgbuilder.py", line 592, in captureStats
    loadavg[0], loadavg[1], loadavg[2], procs[0], procs[1], meminfo["MemAvailable"], \
KeyError: 'MemAvailable'
Parallel build failure - see log for details. Time of failure: Thu Nov 24 10:34:28 CST 2022
make: *** [Makefile:10: image] Error 1

Ntemis, 2022-11-24:
DISTRO=Lakka PROJECT=Generic ARCH=x86_64 DEVICE=x11 make image

greatliu, 2022-11-24:
> DISTRO=Lakka PROJECT=Generic ARCH=x86_64 DEVICE=x11 make image

I tried, but it shows the same error.

DISTRO=Lakka PROJECT=Generic ARCH=x86_64 DEVICE=x11 make image
./scripts/image mkimage

 =================================================================================
 Configuration for Lakka (community)
 =================================================================================

 Buildsystem configuration:
 ======================================================
 - CPU:                                  x86-64
 - Kernel Architecture:                  x86
 - Userland Architecture:                x86_64
 - CPU features:                         64bit cmov cx8 fpu fxsr mmx no_sahf osfxsr sce sse sse2
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:         yes
 - MOLD (Modern Linker) Support:
 - Default Linker:
 - LLVM support:                         yes
 - DEBUG:                                no
 - CFLAGS:                               -march=x86-64 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG
 - LDFLAGS:                               -Wl,--as-needed -fuse-ld=bfd

 Misc. hardware configuration:
 ======================================================
 - ALSA support:                         yes
 - Pulseaudio support:                   no
 - Bluetooth support:                    yes
 - Include driver:                       RTL8192DU
 - Include driver:                       RTL8812AU
 - Include driver:                       bcm_sta
 - Include firmware:                     misc-firmware
 - Include firmware:                     wlan-firmware
 - Include firmware:                     dvb-firmware
 - Include firmware:                     iwlwifi-firmware

 Misc. Filesystems:
 ======================================================
 - Swap Support:                         yes
   - Swapfile default size:              128
 - Install HFS Tools:                    yes

 Network service configuration:
 ======================================================
 - Avahi (Zeroconf) support:             yes
 - NFS mounting support:                 yes
 - SAMBA mounting support:               yes
 - SAMBA server support:                 yes
 - SFTP server support:                  yes
 - OpenVPN support:                      yes
 - WireGuard support:                    yes

 Graphic configuration:
 ======================================================
 - Graphic Drivers:                      crocus i915 iris nvidia nvidia-legacy r300 r600 radeonsi virtio vmware
 - Display Server:                       x11
 - X.Org Graphic Drivers:                xf86-video-amdgpu xf86-video-ati xf86-video-intel xf86-video-nvidia xf86-video-nvidia-legacy xf86-video-vmware
 - X.Org Composite support:              yes
 - Window Manager / Compositor:          fluxbox
 - OpenGL (GLX) support (provider):      yes (mesa)
 - OpenGL ES support (provider):         no
 - Vulkan API support (provider):        no

 Video Acceleration configuration:
 ======================================================
 - VA-API Support:                       yes
 - VDPAU Support:                        yes

 OS configuration:
 ======================================================
 - OEM Support:                          no
 - Default ROOT Password:                root
 - Bootloader:                           syslinux
 - UDevil support:                       yes
 - Installer support:                    yes
 - Include package:                      rsync
 - Include package:                      xpadneo

 Lakka configuration:
 ======================================================
  Including optional packages:
                                retroarch_assets
                                retroarch_overlays
                                retroarch_joypad_autoconfig
                                libretro_database
                                core_info
 ======================================================
  Including 128 libretro core(s)

 =================================================================================
 End Configuration for Lakka
 =================================================================================

Packages loaded : 1090
Build trigger(s): 1 [image]
Package steps   : 423

Traceback (most recent call last):
  File "scripts/pkgbuilder.py", line 918, in <module>
    result = builder.build()
  File "scripts/pkgbuilder.py", line 453, in build
    job = self.getCompletedJob()
  File "scripts/pkgbuilder.py", line 559, in getCompletedJob
    job = self.complete.get(block=True, timeout=self.captureStats(finished=False))
  File "scripts/pkgbuilder.py", line 592, in captureStats
    loadavg[0], loadavg[1], loadavg[2], procs[0], procs[1], meminfo["MemAvailable"], \
KeyError: 'MemAvailable'
Parallel build failure - see log for details. Time of failure: Thu Nov 24 11:29:33 CST 2022
make: *** [Makefile:10: image] Error 1

ciberfred, 2023-01-29:
seems a python error, is in 'pip list' you get meminfo ?
if not try install 'pip install meminfo' and retry the build...

ToKe79, 2023-01-29:
@greatliu hi, the build works (at least now), you can also test this image: https://nightly.builds.lakka.tv/members/vudiq/x11.x86_64/
I did not test the image yet, let us know what works and what does not. thanks!

greatliu, 2023-02-01:
I tested the live mode.
It just shows a black screen and a mouse icon on the black screen.......

***@***.***
 
From: Tomáš Kelemen
Date: 2023-01-30 05:34
To: libretro/Lakka-LibreELEC
CC: greatliu; Mention
Subject: Re: [libretro/Lakka-LibreELEC] please add x11support (Issue #1766)
@greatliu hi, the build works (at least now), you can also test this image: https://nightly.builds.lakka.tv/members/vudiq/x11.x86_64/
I did not test the image yet, let us know what works and what does not. thanks!

gouchi, 2023-02-04:
I tested [latest dev build](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/Lakka-x11.x86_64-devbuild-v5.x-2023-02-04-0fccb7f.img.gz) and I got this log

```
Feb 04 18:01:25 Lakka systemd[1]: Starting xorg.service...
Feb 04 18:01:26 Lakka xorg-launch[1061]: Starting Xorg server with: "/usr/bin/Xorg -nolisten tcp vt01 -s 0 -noreset -allowMouseOpenFail -config /etc/X11/xorg-i915.conf"
Feb 04 18:01:26 Lakka xorg-launch[1061]: X.Org X Server 1.21.1.6
Feb 04 18:01:26 Lakka xorg-launch[1061]: X Protocol Version 11, Revision 0
Feb 04 18:01:26 Lakka xorg-launch[1061]: Current Operating System: Linux Lakka 6.1.9 #1 SMP Sat Feb  4 14:33:54 CET 2023 x86_64
Feb 04 18:01:26 Lakka xorg-launch[1061]: Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=0402-3741 disk=UUID=c64c8ec5-43e6-44c9-a57f-bcf9114bda77 portable quiet
Feb 04 18:01:26 Lakka xorg-launch[1061]:  
Feb 04 18:01:26 Lakka xorg-launch[1061]: Current version of pixman: 0.42.2
Feb 04 18:01:26 Lakka xorg-launch[1061]:         Before reporting problems, check https://libreelec.tv/
Feb 04 18:01:26 Lakka xorg-launch[1061]:         to make sure that you have the latest version.
Feb 04 18:01:26 Lakka xorg-launch[1061]: Markers: (--) probed, (**) from config file, (==) default setting,
Feb 04 18:01:26 Lakka xorg-launch[1061]:         (++) from command line, (!!) notice, (II) informational,
Feb 04 18:01:26 Lakka xorg-launch[1061]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Feb 04 18:01:26 Lakka xorg-launch[1061]: (==) Log file: "/var/log/Xorg.0.log", Time: Sat Feb  4 18:01:26 2023
Feb 04 18:01:26 Lakka xorg-launch[1061]: (++) Using config file: "/etc/X11/xorg-i915.conf"
Feb 04 18:01:26 Lakka xorg-launch[1061]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Feb 04 18:01:26 Lakka avahi-daemon[761]: Server startup complete. Host name is Lakka.local. Local service cookie is 3505585814.
Feb 04 18:01:27 Lakka avahi-daemon[761]: Service "Lakka" (/etc/avahi/services/ssh.service) successfully established.
Feb 04 18:01:27 Lakka avahi-daemon[761]: Service "Lakka" (/etc/avahi/services/sftp-ssh.service) successfully established.
Feb 04 18:01:27 Lakka systemd[1]: xorg.service: Got notification message from PID 1061, but reception only permitted for main PID 1057
Feb 04 18:01:27 Lakka systemd[1]: Started xorg.service.
Feb 04 18:01:27 Lakka systemd[1]: Started windowmanager.service.
Feb 04 18:01:27 Lakka systemd[1]: Reached target graphical.target.
Feb 04 18:01:27 Lakka systemd[1]: Starting retroarch-autostart.service...
Feb 04 18:01:27 Lakka systemd[1]: Starting systemd-update-utmp-runlevel.service...
Feb 04 18:01:27 Lakka systemd[1]: systemd-update-utmp-runlevel.service: Deactivated successfully.
Feb 04 18:01:27 Lakka systemd[1]: Finished systemd-update-utmp-runlevel.service.
Feb 04 18:01:27 Lakka sh[1070]: /bin/sh: can't open '/storage/.config/autostart.sh': No such file or directory
Feb 04 18:01:27 Lakka systemd[1]: Finished retroarch-autostart.service.
Feb 04 18:01:27 Lakka systemd[1]: retroarch.service: Failed to load environment files: No such file or directory
Feb 04 18:01:27 Lakka systemd[1]: retroarch.service: Failed to run 'start-pre' task: No such file or directory
Feb 04 18:01:27 Lakka systemd[1]: retroarch.service: Failed with result 'resources'.
Feb 04 18:01:27 Lakka systemd[1]: Failed to start retroarch.service.
Feb 04 18:01:27 Lakka systemd[1]: Reached target retroarch.target.
```

ToKe79, 2023-02-04:
the latest dev build should have fixes that address this - I had to replace the systemd dependencies (substitute kodi for retroarch).

ToKe79, 2023-02-04:
but you were running the build... strange, I will check again... another issue was missing env for DISPLAY, which should have been added.

ToKe79, 2023-02-04:
should be fixed with 575eef7f376ba70cc4ce2ace71a52259346425bf

ToKe79, 2023-02-04:
on the first boot (after filesystem set up) it complains that xorg failed to start due to windowmanager failure, but eventually it starts in to retroarch, subsequent boots do not have this issue. I had to manually select audio device (using left/right d-pad/joystick on Settings - Audio - Output - Device).

I tested with onboard ATI/AMD HD3000 GPU.

gouchi, 2023-02-14:
I confirm that [latest nightly build](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/) is working with X11.

@greatliu Can you make a test ? So that if it is working we can close this issue.

Thank you.

gouchi, 2023-12-19:
Closing this issue as [X11 build is available](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/).