libretro/Lakka-LibreELEC #1768: Lakka 4.2 ODROID-C4 - WiFi Module 5BK not working

mvrk69, 2022-10-26:
Hi,

I've flashed Lakka 4.2 on my odroid-c4, but wifi doesn't work.

I'm using WiFi Module 5BK (https://www.hardkernel.com/shop/wifi-module-5bk)

Seems firmware is missing:

[  111.538776] usb 1-1.1: USB disconnect, device number 3
[  115.138360] usb 1-1.1: new high-speed USB device number 5 using xhci-hcd
[  115.286921] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[  115.286949] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  115.286961] usb 1-1.1: Product: 802.11ac NIC
[  115.286970] usb 1-1.1: Manufacturer: Realtek
[  115.286979] usb 1-1.1: SerialNumber: 123456
[  115.381190] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[  115.382176] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  115.382201] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[  115.382526] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821c_fw.bin failed with error -2
[  115.382553] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8821c_fw.bin not found
[  115.415154] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[  115.416144] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  115.416166] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[  115.416356] bluetooth hci0: Direct firmware load for rtl_bt/rtl8821c_fw.bin failed with error -2
[  115.416369] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8821c_fw.bin not found

gouchi, 2022-11-23:
Can you try this [testing image](https://nightly.builds.lakka.tv/2022-11-19/AMLGX.aarch64/Lakka-AMLGX.aarch64-nightly-v4.x-20221119-5a6e401-odroid-c4.img.gz) to check if your issue fixed ?  Thanks to @by-gnome

Ntemis, 2023-01-20:
Issue is fixed

closed 2023-01-20 by Ntemis

mvrk69, 2023-01-30:
Doesn't seem to be working, seems now only detects the bluetooth part:

[  171.962995] usb 1-1.1: USB disconnect, device number 3
[  178.120613] usb 1-1.1: new high-speed USB device number 5 using xhci-hcd
[  178.273258] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[  178.273290] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  178.273304] usb 1-1.1: Product: 802.11ac NIC
[  178.273315] usb 1-1.1: Manufacturer: Realtek
[  178.273325] usb 1-1.1: SerialNumber: 123456
[  178.366249] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[  178.367211] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  178.367232] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[  178.367640] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_config.bin
[  178.367942] Bluetooth: hci0: RTL: cfg_sz 10, total sz 31990
[  178.819218] Bluetooth: hci0: RTL: fw version 0x829a7644

Lakka:~ # lsusb -tv
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 2109:0817 VIA Labs, Inc.
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:2817 VIA Labs, Inc.
        |__ Port 1: Dev 5, If 0, Class=Wireless, Driver=btusb, 480M
            ID 0bda:c820 Realtek Semiconductor Corp.
        |__ Port 1: Dev 5, If 1, Class=Wireless, Driver=btusb, 480M
            ID 0bda:c820 Realtek Semiconductor Corp.
        |__ Port 1: Dev 5, If 2, Class=Vendor Specific Class, Driver=, 480M
            ID 0bda:c820 Realtek Semiconductor Corp.

mvrk69, 2023-02-02:
I will open new issue

PR #1775 (2022-11-13): amlogic: add rtl8821c support to kernel-firmware
This should fix #1768