libretro/Lakka-LibreELEC #1769: Enable savestate in Hatari config

schauveau, 2022-10-29:
I recently installed Lakka 4.2 on a rPi4 and I noticed that save/load state is missing when using the Hatari core.
I was a bit surprised because that was working during my previous experiments with both Retropie and a manually built retroarch.

As far as I can tell, the feature is present in `hatari_libretro.so` but it is not enabled in `hatari_libretro.info`

I was able to make it work by cloning the hatari core. 
Copy `/usr/lib/libretro/hatari_libretro.*` to `/storage/cores/hatari_save_libretro.*` 
Then edit the `hatari_save_libretro.info` with 
 
    display_name = "Atari - ST/STE/TT/Falcon (Hatari with save) "
    savestate = "true"
    savestate_features = "serialized"

I tested the 'clone' core on a few games. Saving and loading appears to work fine. If there are no major issues with the feature, it should be enabled by default in the next Lakka release

Ntemis, 2022-10-29:
its already there 
https://github.com/libretro/libretro-core-info/blob/master/hatari_libretro.info#L18-L19

schauveau, 2022-10-29:
Ha ok! I could not figure out where the info files were coming from. Thanks.

closed 2022-10-29 by schauveau