libretro/Lakka-LibreELEC #1770: Generic usb controller not working properly in lakka

hugod23, 2022-11-04:
Hello, I'm using raspberry pi zero 1.3 (version without wifi and bluetooth) to emulate games with Lakka 4.2, but I'm having problems with the generic chinese usb controller I'm using, when I plug it into the board it shows as recognized "Dragonrise N64", but only the directional pads work. buttons 1,2,3,4 and others don't work, another thing when I disconnect it from the usb port a message appears that I disconnected the controller, but it appears the name "DragonRise Inc. Generic USB Joystick".

Can someone help me? I'm a newbie starting with these emulation systems

[image]

Ntemis, 2022-11-04:
lsusb

by-gnome, 2022-11-10:
Go to the `Input Settings` using a keyboard and `Bind all` bottons on your joypad.

Graham15, 2023-06-19:
I have a similar joypad (looks exactly the same apart from the d-pad) but my problem is that button 2 isn't recognized (during auto config) but the other buttons are. I'm not sure what you mean by "Bind all". Is that an option in the menu? (maybe it has been renamed to Set All Controls now). Like by-gnome said, using a keyboard gets around this but would have been nice to setup fully with the joypad.

:)

gouchi, 2023-12-19:
Can you make a test with [Lakka 4.3](https://le-builds.lakka.tv/) and [Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/) if you reproduce the issue ?

Thank you.

ToKe79, 2024-04-04:
no response

additional info: some gamepads have a switch to toggle DirectInput and XInput, try switching to different operation mode. also some gamepad vendors do not implement the protocol/buttons/etc correctly / according to expected behavior of the Linux kernel, so those are hard to support. maybe usb quirks (?) could help fix the behavior.

closed 2024-04-04 by ToKe79