libretro/Lakka-LibreELEC #1773: Vulkan vidoe driver hangs/crashes when configured for use in cores and menu

auxfam, 2022-11-10:
If you are sure that it's a bug in Lakka, please answer the following questions:
> Which version of Lakka are you using?
nightlies latest: https://nightly.builds.lakka.tv/2022-10-17/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221017-1fa967d.img.gz

> What system hardware are you using?
PN51, AMD APU 5700u.  Zen2+Vega8

> What did you do?
1. boot up swanstation title (any)
2. create per game config
3. set video driver in core options from auto to 'vulkan'
4. close game
5. select from history and run game
6. crashes back to menu after a couple of seconds

> What did you expect to happen?
game should start

> What happened instead?
crashed back to menu

* note
an easier way to recreate this is to simply
1. boot into retroarch
2. set the GLOBAL video driver to 'vulkan'
3. reboot

it will hang at the flower boot icon
then you have to ssh onto the box and edit or delete `.config/retroarch/retroarch.cfg` and reboot

** to fix this I had to rollback a number of versions specifically
https://nightly.builds.lakka.tv/2022-10-02/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221002-e8a1bb6.img.gz

and yes this occurs on other vulkan supported cores.  luckily the ssh daemon does start up so sshhing onto the box and editing cfgs back to 'gl' or 'glcore' stops the crashing.  all points toward a change in vulkan I suspect

auxfam, 2022-11-15:
has 22.1.7 been deployed on master and in the latest builds?  or is that pending? (latest builds from nightlies still exhibit this problem so I'll assume the latter)

Ntemis, 2022-11-15:
Hmm well did you try any other vulkan core?

auxfam, 2022-11-15:
based on https://nightly.builds.lakka.tv/2022-11-11/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221111-7bb6c38.img.gz

1. snex9x, genesisgx, flip to vulkan driver on per game title.   crashes to menu
2. swanstation per game option, set video (in core options) from auto - > vulkan, save as per game.  close and run, crashes to menu
3. final test.  no core loaded just set video driver to vulkan and reboot.  hangs on flower icon on boot.  (ssh server is still up)

Ntemis, 2022-11-15:
dmesg log please
also stop retroarch and start it again with -v and give me the log of the craash

auxfam, 2022-11-15:
[dmesg.log.zip](https://github.com/libretro/Lakka-LibreELEC/files/10012208/dmesg.log.zip)
see above this is after

1. set video driver global to vulkan
2. reboot (not restart) from menu
3. freeze at flower icon

can you elaborate on how you want me to test ```also stop retroarch and start it again with -v ```

auxfam, 2022-11-15:
ok not sure if I did this right
[log-2022-11-16-13.25.37.zip](https://github.com/libretro/Lakka-LibreELEC/files/10012639/log-2022-11-16-13.25.37.zip)

1. boot lakka
2. load game on dolphin set to use vulkan video driver
3. watch it crash back to menu
4. ssh onto box
5. `createlog` command
6. dumped log to zip.  (see attached)

if I am misusing this let me know

Ntemis, 2022-11-15:
kernel and gpu driver look fine so i need to see where retroarch is crashing and the logs dont have much to work on

Ntemis, 2022-11-15:
i found the issue

Ntemis, 2022-11-15:
@auxfam compile ~head and i let you close this :)
Thank you for the report!

gouchi, 2022-11-17:
@auxfam you can try this [nightly build](https://nightly.builds.lakka.tv/2022-11-16/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221116-96c6653.img.gz.sha256) which contains the fix.

auxfam, 2022-11-17:
sorry this one took longer since
1. something changed in the partition requirements so I had to switch to another ssd and install fresh
2. after doing so teh same issue persists
3. this time I think I can see logs though

`[INFO] RetroArch 1.10.3 (Git 9b282aa)
[INFO] === Build =======================================
[INFO] CPU Model Name: AMD Ryzen 7 5700U with Radeon Graphics
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Aug  4 2022
[INFO] Version: 1.10.3
[INFO] Git: 9b282aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[WARN] Failed to bind API (#9, version 1.0) on context driver "kms".
[INFO] [Vulkan]: Found vulkan context: "null".
[INFO] [Vulkan]: Detecting screen resolution: 320x240.
[INFO] [Vulkan]: Using resolution 320x240.
[ERROR] [Vulkan]: Failed to get context data.
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
`
above is when it hanges on boot after setting display driver to vulkan.

attached is switching to the bulkan renderer in Mupens options, close and start rom/game
[retroarch__2022_11_18__16_30_17.log](https://github.com/libretro/Lakka-LibreELEC/files/10033042/retroarch__2022_11_18__16_30_17.log)

auxfam, 2022-11-17:
excuse the log file names (I messaed up teh bios date when reinstalling/ fixed).  same behaviour though
```
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: SET_HW_RENDER, context type: vulkan.
[INFO] Requesting Vulkan context.
[INFO] Reached end of SET_HW_RENDER.
[INFO] [Environ]: SET_HW_RENDER_CONTEXT_NEGOTIATION_INTERFACE.
[libretro INFO] mupen64plus: Using full mem base
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro DEBUG] mupen64plus: [EmuThread] M64CMD_ROM_OPEN
[libretro INFO] mupen64plus: Goodname: 007 - The World is Not Enough (U) [!]
[libretro INFO] mupen64plus: Name: TWINE
[libretro INFO] mupen64plus: MD5: 9D58996A8AA91263B5CD45C385F45FE4
[libretro INFO] mupen64plus: CRC: 033F4C13 319EE7A7
[libretro INFO] mupen64plus: Imagetype: .v64 (byteswapped)
[libretro INFO] mupen64plus: Rom size: 33554432 bytes (or 32 Mb or 256 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 144B
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 374F
[libretro INFO] mupen64plus: Country: USA
[libretro INFO] mupen64plus: PC = 80000400
[libretro INFO] mupen64plus: Save type: 5
[libretro DEBUG] mupen64plus: [EmuThread] M64CMD_ROM_GET_HEADER
[INFO] [State]: Found auto save state in "/storage/savestates/Mupen64Plus-Next/007 - The World Is Not Enough (USA).state.auto".
[INFO] [State]: Auto-loading save state from "/storage/savestates/Mupen64Plus-Next/007 - The World Is Not Enough (USA).state.auto" succeeded.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[INFO] [Vulkan]: App: paraLLEl-RDP (version 0)
[INFO] [Vulkan]: Engine: Granite (version 0)
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[INFO] [Vulkan]: App: paraLLEl-RDP (version 0)
[INFO] [Vulkan]: Engine: Granite (version 0)
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[WARN] Failed to bind API (#9, version 1.0) on context driver "kms".
[INFO] [Vulkan]: Found vulkan context: "null".
[INFO] [Vulkan]: Detecting screen resolution: 320x240.
[INFO] [Vulkan]: Using resolution 320x240.
[ERROR] [Vulkan]: Failed to get context data.
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading game..
[libretro INFO] mupen64plus: Rom closed.
[INFO] [Core]: Unloading core..
[libretro DEBUG] mupen64plus: [EmuThread] M64CMD_EXECUTE
```

let me know if there is anything else useful I can test with, else I'll revert back to a previous version after a while

Ntemis, 2022-11-18:
Investigating further hold on

Ntemis, 2022-11-18:
@auxfam this took too long but i think this time i nailed it.
Let me know

Ntemis, 2022-11-19:
https://nightly.builds.lakka.tv/2022-11-19/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221119-5a6e401.img.gz

auxfam, 2022-11-21:
a bit different in terms of the logs but still fails
swanstation:```[INFO] [Environ]: GET_VARIABLE: duckstation_CDROM.LoadImagePatches = "true"
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_VARIABLE: duckstation_BIOS.PathNTSCJ = "scph5500.bin"
[INFO] [Environ]: SET_MESSAGE: CD-ROM read speedup set to 3x (effective speed 6x). This may result in instability.
[INFO] [Environ]: SET_MESSAGE: CD-ROM seek speedup set to 4x. This may result in instability.
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[INFO] [Environ]: SET_MESSAGE: Memory card at '/storage/savefiles/SwanStation/Zoku_Mikagura_Shoujo_Tanteidan_-_Kanketsuhen_-jp-__-Disc_1-__1.mcd' could not be read, formatting.
[INFO] [Environ]: SET_MESSAGE: Saved memory card to '/storage/savefiles/SwanStation/Zoku_Mikagura_Shoujo_Tanteidan_-_Kanketsuhen_-jp-__-Disc_1-__1.mcd'
[INFO] [Environ]: GET_VARIABLE: duckstation_GPU.Renderer = "Vulkan"
[INFO] [Environ]: GET_PREFERRED_HW_RENDER, video driver name: gl.
[INFO] [Environ]: GET_PREFERRED_HW_RENDER - Context callback set to RETRO_HW_CONTEXT_OPENGL.
[INFO] [Environ]: SET_HW_RENDER, context type: vulkan.
[INFO] Requesting Vulkan context.
[INFO] Reached end of SET_HW_RENDER.
[INFO] [Environ]: SET_HW_RENDER_CONTEXT_NEGOTIATION_INTERFACE.
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 44234.67 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Video]: "vulkan" saved as cached driver.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[WARN] Failed to bind API (#9, version 1.0) on context driver "kms".
[INFO] [Vulkan]: Found vulkan context: "null".
[INFO] [Vulkan]: Detecting screen resolution: 320x240.
[INFO] [Vulkan]: Using resolution 320x240.
[ERROR] [Vulkan]: Failed to get context data.
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading game..
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved core options file to "/storage/.config/retroarch/config/SwanStation/SwanStation.opt".
[INFO] [Overrides]: Configuration overrides unloaded, original configuration restored.
[ERROR] failed_to_start_audio_driver
```

flycast ```[INFO] [Environ]: GET_VARIABLE: reicast_threaded_rendering = "disabled"
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: GET_VARIABLE: reicast_alpha_sorting = "per-triangle (normal)"
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[libretro INFO] shell/libretro/libretro.cpp:2020 N[RENDERER]: retro_get_system_av_info: Res=960
[INFO] [Audio]: Set audio input rate to: 44101.76 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[INFO] [Vulkan]: App: Flycast (version 1)
[INFO] [Vulkan]: Engine: Flycast (version 1)
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[INFO] [Vulkan]: App: Flycast (version 1)
[INFO] [Vulkan]: Engine: Flycast (version 1)
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[... 62 lines cut ...]
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[INFO] [Font]: Using font rendering backend: "freetype".
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "Power Button" (/dev/input/event2).
[INFO] [udev]: Keyboard #1: "Video Bus" (/dev/input/event3).
[INFO] [udev]: Keyboard #2: "Power Button" (/dev/input/event0).
[INFO] [udev]: Keyboard #3: "Sleep Button" (/dev/input/event1).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA]: Using signed 16-bit format.
[INFO] [ALSA]: Period size: 768 frames
[INFO] [ALSA]: Buffer size: 3072 frames
[INFO] [ALSA]: Can pause: yes.
[INFO] [Display]: Found display driver: "gl".
[INFO] [SRAM]: SRAM will not be saved.```

Ntemis, 2022-11-22:
Does retroarch works fine with video_driver = "vulkan" ?

auxfam, 2022-11-23:
1. drivers -> video -> switch from `gl` to `vulkan`
2. reboot
3. hangs on flower icon
4.  log
5. ```[INFO] RetroArch 1.10.3 (Git 9b282aa)
[INFO] === Build =======================================
[INFO] CPU Model Name: AMD Ryzen 7 5700U with Radeon Graphics
[INFO] Capabilities:  MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Aug  4 2022
[INFO] Version: 1.10.3
[INFO] Git: 9b282aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[ERROR] [Vulkan]: Could not find instance extensions. Will attempt without them.
[ERROR] Failed to create Vulkan instance (-9).
[ERROR] [Vulkan]: Failed to create Vulkan context.
[WARN] Failed to bind API (#9, version 1.0) on context driver "kms".
[INFO] [Vulkan]: Found vulkan context: "null".
[INFO] [Vulkan]: Detecting screen resolution: 320x240.
[INFO] [Vulkan]: Using resolution 320x240.
[ERROR] [Vulkan]: Failed to get context data.
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
```
6. ssh onto box and exit `.config/retroarch/retroarch.cfg`  and change `video_driver = "vulkan"` to `video_driver = "gl"`
7. reboot ( a la shutdown -r)
8. goes to menu (normal)

so still issues with setting the vulkan video driver I am afraid

auxfam, 2022-12-19:
jsut a quick update.  Latest nightly https://nightly.builds.lakka.tv/2022-12-19/Generic.x86_64/Lakka-Generic.x86_64-nightly-v4.x-20221219-8b6e706.img.gz 
still same issue,  flycast core (but same ofr others)
`[INFO] RetroArch 1.14.0 (Git ad89b0c)
[INFO] === Build =======================================
[INFO] CPU Model Name: AMD Ryzen 7 5700U with Radeon Graphics
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Dec 16 2022
[INFO] Version: 1.14.0
[INFO] Git: ad89b0c
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/flycast_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
[INFO] [Overrides]: Game-specific overrides found at "/storage/.config/retroarch/config/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.cfg".
[INFO] [Overrides]: Game-specific overrides stacking on top of previous overrides.
[INFO] [Config]: Loading config: "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Config]: Appending config: "/storage/.config/retroarch/config/Flycast/Flycast.cfg".
[INFO] [Config]: Appending config: "/storage/.config/retroarch/config/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2_INTL.
[INFO] [Core]: Game-specific core options found at "/storage/.config/retroarch/config/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.opt".
[DEBUG] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_UPDATE_DISPLAY_CALLBACK.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO]    Controller port: 1
[INFO]       Controller (ID: 1)
[INFO]       Arcade Stick (ID: 1025)
[INFO]       Keyboard (ID: 3)
[INFO]       Mouse (ID: 2)
[INFO]       Light Gun (ID: 4)
[INFO]       Twin Stick (ID: 513)
[INFO]       Saturn Twin-Stick (ID: 769)
[INFO]    Controller port: 2
[INFO]       Controller (ID: 1)
[INFO]       Arcade Stick (ID: 1025)
[INFO]       Keyboard (ID: 3)
[INFO]       Mouse (ID: 2)
[INFO]       Light Gun (ID: 4)
[INFO]       Twin Stick (ID: 513)
[INFO]       Saturn Twin-Stick (ID: 769)
[INFO]    Controller port: 3
[INFO]       Controller (ID: 1)
[INFO]       Arcade Stick (ID: 1025)
[INFO]       Keyboard (ID: 3)
[INFO]       Mouse (ID: 2)
[INFO]       Light Gun (ID: 4)
[INFO]       Twin Stick (ID: 513)
[INFO]       Saturn Twin-Stick (ID: 769)
[INFO]    Controller port: 4
[INFO]       Controller (ID: 1)
[INFO]       Arcade Stick (ID: 1025)
[INFO]       Keyboard (ID: 3)
[INFO]       Mouse (ID: 2)
[INFO]       Light Gun (ID: 4)
[INFO]       Twin Stick (ID: 513)
[INFO]       Saturn Twin-Stick (ID: 769)
[INFO] [Remaps]: Core-specific remap found at "/storage/remappings/Flycast/Flycast.rmp".
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[libretro INFO] shell/libretro/libretro.cpp:309 N[BOOT]: retro_init
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: SET_KEYBOARD_CALLBACK.
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[libretro INFO] core/hw/mem/_vmem.cpp:494 N[VMEM]: Info: nvmem is enabled, with addr space of size 4GB
[libretro INFO] core/hw/mem/_vmem.cpp:589 N[VMEM]: BASE 0x7f9c5fff0000 RAM(16 MB) 0x7f9cebff0000 VRAM64(8 MB) 0x7f9ce3ff0000 ARAM(2 MB) 0x7f9ce07f0000
[INFO] [Content]: Content loading skipped. Implementation will load it on its own.
[libretro INFO] shell/libretro/libretro.cpp:1785 N[BOOT]: retro_load_game: /storage/roms/int-games/Sega_-_Dreamcast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.chd
[... 243 lines cut ...]
[INFO] [Environ]: GET_VARIABLE: reicast_texupscale = "N/A"
[INFO] [Environ]: GET_VARIABLE: reicast_texupscale_max_filtered_texture_size = "N/A"
[INFO] [Environ]: GET_VARIABLE: reicast_custom_textures = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_dump_textures = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_fog = "enabled"
[INFO] [Environ]: GET_VARIABLE: rend.PerStripSorting = "N/A"
[INFO] [Environ]: GET_VARIABLE: reicast_delay_frame_swapping = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_widescreen_cheats = "enabled"
[INFO] [Environ]: GET_VARIABLE: reicast_frame_skipping = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_auto_skip_frame = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_threaded_rendering = "enabled"
[INFO] [Environ]: GET_VARIABLE: reicast_anisotropic_filtering = "16"
[INFO] [Environ]: GET_VARIABLE: reicast_texture_filtering = "0"
[INFO] [Environ]: GET_VARIABLE: reicast_pvr2_filtering = "enabled"
[INFO] [Environ]: GET_VARIABLE: reicast_oit_layers = "32"
[INFO] [Environ]: GET_VARIABLE: reicast_native_depth_interpolation = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_hle_bios = "disabled"
[INFO] [Environ]: GET_VARIABLE: reicast_gdrom_fast_loading = "enabled"
[INFO] [Environ]: GET_VARIABLE: reicast_network_output = "disabled"
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved game-specific core options to "/storage/.config/retroarch/config/Flycast/Dead_or_Alive_2_-jp-__-Shokai_Genteiban-_.opt".
[INFO] [Config]: Loading config: "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Overrides]: Configuration overrides unloaded, original configuration restored.
[ERROR] failed_to_start_audio_driver`

Ntemis, 2022-12-23:
@auxfam please test the latest nightly and close if ok, thank you
https://nightly.builds.lakka.tv/2022-12-23/

auxfam, 2022-12-23:
Hi,

SwanStation, Flycast and menu drier reboot all tested and now work
```
[INFO] RetroArch 1.14.0 (Git ad89b0c)
[INFO] === Build =======================================
[INFO] CPU Model Name: AMD Ryzen 7 5700U with Radeon Graphics
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Dec 16 2022
[INFO] Version: 1.14.0
[INFO] Git: ad89b0c
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "AMD RADV RENOIR".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Found GPU at index 0: "AMD RADV RENOIR".
[INFO] [Vulkan]: Using GPU index 0.
[INFO] [Vulkan]: Using semaphores for WSI acquire.
[INFO] [Vulkan]: Using GPU: "AMD RADV RENOIR".
[INFO] [Vulkan]: Queue family 0 supports 1 sub-queues.
[INFO] [Vulkan]: Got 2 swapchain images.
[INFO] [Vulkan]: Using resolution 1920x1080.
[INFO] [Vulkan]: Using RGB565 format.
[INFO] [Vulkan]: Loading stock shader.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [udev]: Keyboard #0: "Power Button" (/dev/input/event2).
[INFO] [udev]: Keyboard #1: "Video Bus" (/dev/input/event3).
[INFO] [udev]: Keyboard #2: "Power Button" (/dev/input/event0).
[INFO] [udev]: Keyboard #3: "Sleep Button" (/dev/input/event1).
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: "null".
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA]: Using signed 16-bit format.
[INFO] [ALSA]: Period size: 768 frames
[INFO] [ALSA]: Buffer size: 3072 frames
[INFO] [ALSA]: Can pause: yes.
[INFO] [Display]: Found display driver: "vulkan".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Config]: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Config]: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Video]: Does not have enough samples for monitor refresh rate estimation. Requires to run for at least 4096 frames.
```

the only exception is the dolphin core  which does not show errors but crashes to menu (gl works fine)
```
[INFO] === Build =======================================
[INFO] CPU Model Name: AMD Ryzen 7 5700U with Radeon Graphics
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSE4 SSE4.2 AES AVX AVX2
[INFO] Built: Dec 16 2022
[INFO] Version: 1.14.0
[INFO] Git: ad89b0c
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[... 66 lines cut ...]
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO]    Controller port: 1
[INFO]       GameCube Controller (ID: 1)
[INFO]    Controller port: 2
[INFO]       GameCube Controller (ID: 1)
[INFO]    Controller port: 3
[INFO]       GameCube Controller (ID: 1)
[INFO]    Controller port: 4
[INFO]       GameCube Controller (ID: 1)
[INFO] [SRAM]: Skipping SRAM load.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Environ]: GET_VARIABLE: dolphin_mixer_rate = "32000"
[INFO] [Audio]: Set audio input rate to: 32029.64 Hz.
[INFO] [Video]: Set video size to: fullscreen.
[INFO] [Video]: Using HW render, vulkan driver forced.
[INFO] [Vulkan]: Vulkan dynamic library loaded.
[INFO] [Vulkan]: Found vulkan context: "khr_display".
[INFO] [Vulkan]: Detecting screen resolution: 0x0.
[INFO] [Vulkan]: Found GPU at index 0: "AMD RADV RENOIR".
[INFO] [Vulkan]: Using GPU index 0.
```
HOWEVER the dolphin core has been doing this for quite a number of revisions now (from last stable)  so I think it is core related (hasn't been updaed in quite a while now)  so I think this ticket can be closed.

Thank you again for looking into this

closed 2022-12-23 by auxfam

Ntemis, 2022-12-24:
hmm vulkan can not set a display resolution
Detecting screen resolution: 0x0.