libretro/Lakka-LibreELEC #1782: Iwlwifi driver not loading

Dehaj85, 2022-11-22:
iwlwifi driver is not loading when Lakka is installed. Works with TP-Link N150 USB Dongle. Iwlwifi driver should be incorporated with kernel 4.1 and up. Driver does not display when checking through command line terminal.

This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

gouchi, 2022-11-23:
On which architecture ?

Because it is already [enabled for some platform](https://github.com/libretro/Lakka-LibreELEC/search?q=CONFIG_IWLWIFI).

Thank you.

Dehaj85, 2022-11-23:
It is an x86 system with an intel 8260 combo NIC with BT.  confirmed the NIC is working with a live version of unbuntu and windows so the NIC is not dead.

gouchi, 2022-11-24:
Can you try to provide some with [log](https://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) with `createlog` ?

Thank you.

Ntemis, 2023-01-20:
No requested logs provided closing down, feel free to reopen

closed 2023-01-20 by Ntemis