libretro/Lakka-LibreELEC #1783: RetroArch does not start with error message about MESA

panam510, 2022-11-24:
There seems to be a problem with the GPU and I can't get RetroArch to run on my old netbook.

The log outputs the following error:
```
MESA: error: Out of instructions
MESA: error: Exceeded max ALU instructions
```
```
MESA: error: bad opcode BGNLOOP (99)
```

- Which version of Lakka are you using?
- - Generic.i386-devel-20220621002118-be8926b
- What system hardware are you using?
- - An old x86 netbook with Intel Atom N280 CPU and 945GSE graphics (GIGABYTE M1022X)
- What did you do?
- - I installed it to a USB flash drive and booted in live mode. By the way, my netbook had ACPI problems, so I gave it the boot option of `acpi=off`.
- What did you expect to happen?
- - I expected normal startup, but it did not start up, so I changed the menu driver from `xmb` to `rgui`.
- What happened instead?
- - The message was different, but still did not start.

[log-2022-11-24-15.45.38.zip](https://github.com/libretro/Lakka-LibreELEC/files/10081529/log-2022-11-24-15.45.38.zip)
[log_with_rgui.txt](https://github.com/libretro/Lakka-LibreELEC/files/10081531/log_with_rgui.txt)

gouchi, 2022-12-13:
I am afraid your GPU is too  old with the current graphic stack we are using.

We had issues with this VGA controller `8086:27ae`  already [1](https://github.com/libretro/Lakka-LibreELEC/issues/121) [2](https://github.com/libretro/Lakka-LibreELEC/issues/979) but we did not succeed to fix it.

panam510, 2022-12-23:
I understood that those issues were closed without being fixed. Thanks for the information.
Just to be sure, I tried adding `menu_shader_pipeline = "0"` and `video_monitor_index = "1"` to `retroarch.cfg` as well as the issues you presented, but it still did not start.

Ntemis, 2023-01-20:
GPU is unsupported on current mesa driver, cant be fixed

closed 2023-01-20 by Ntemis