libretro/Lakka-LibreELEC #1796: Missing GMA500 DRI drivers

psergiu, 2022-12-16:
- Which version of Lakka are you using?

Generic.i386 4.2
Generic.i386-nightly-v4.x-20220621-be8926b

- What system hardware are you using?

NEC Infrontia Corporation GRAPHIC ORDER TERMINAL
Tablet PC with touch-screen - perfect for ScummVM games.

- What did you do?

Booted from USB stick with "live"

- What did you expect to happen?

Lakka in it's full glory

- What happened instead?

Stuck on flower logo

The device has a Intel GMA500 graphics card and it seems that some DRI drivers are missing from Lakka:
/usr/lib/dri/gma500_dri.so
/usr/lib/dri/kms_swrast_dri.so
/usr/lib/dri/swrast_dri.so

Please see attached logs from both 4.2 and latest nightly.

Thank you !

[4.2_log-2022-12-16-16.26.41.zip](https://github.com/libretro/Lakka-LibreELEC/files/10247505/4.2_log-2022-12-16-16.26.41.zip)
[nightly_log-2022-12-16-16.39.51.zip](https://github.com/libretro/Lakka-LibreELEC/files/10247506/nightly_log-2022-12-16-16.39.51.zip)

psergiu, 2022-12-17:
Excerpt from Xorg.0.log of a live image of Puppy Linux booted on the same machine, with X (fbdev) working fine:
Will try other Linux distros and will post logs from one that uses gma500_dri

```
[    78.032]
X.Org X Server 1.19.6
Release Date: 2017-12-20
[    78.032] X Protocol Version 11, Revision 0
[    78.032] Build Operating System: Linux 4.4.0-169-generic i686 Ubuntu
[    78.032] Current Operating System: Linux puppypc32222 4.9.163-lxpup-32-pae #1 SMP Thu Mar 14 15:41:19 GMT 2019 i686
[    78.032] Kernel command line: BOOT_IMAGE=vmlinuz pmedia=cd initrd=initrd.gz
[    78.032] Build Date: 14 November 2019  06:19:18PM
[    78.032] xorg-server 2:1.19.6-1ubuntu4.4 (For technical support please see http://www.ubuntu.com/support)
[    78.032] Current version of pixman: 0.34.0
[    78.032]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    78.032] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    78.032] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Dec 16 16:59:09 2022
[    78.038] (==) Using config file: "/etc/X11/xorg.conf"
[    78.039] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    78.039] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    78.132] (==) ServerLayout "X.org config"
[    78.132] (**) |-->Screen "Screen0" (0)
[    78.132] (**) |   |-->Monitor "Monitor0"
[    78.133] (**) |   |-->Device "Card0"
[    78.133] (**) Option "DontZap" "false"
[    78.134] (**) Option "RandR" "on"
[    78.134] (**) Option "AutoAddDevices" "true"
[    78.134] (**) Automatically adding devices
[    78.134] (==) Automatically enabling devices
[    78.134] (==) Automatically adding GPU devices
[    78.134] (==) Automatically binding GPU devices
[    78.134] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    78.134] (WW) The directory "/usr/share/X11/fonts/Type1/" does not exist.
[    78.134]    Entry deleted from font path.
...
[    78.137] (==) ModulePath set to "/usr/lib/xorg/modules"
[    78.137] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    78.137] (II) Loader magic: 0x6b7020
[    78.137] (II) Module ABI versions:
[    78.137]    X.Org ANSI C Emulation: 0.4
[    78.137]    X.Org Video Driver: 23.0
[    78.137]    X.Org XInput driver : 24.1
[    78.137]    X.Org Server Extension : 10.0
[    78.139] (--) using VT number 4

[    78.140] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    78.142] (II) xfree86: Adding drm device (/dev/dri/card0)
[    78.175] (--) PCI:*(0:0:2:0) 8086:4102:1462:a380 rev 4, Mem @ 0xff200000/1048576, 0xd0000000/268435456, 0xcffc0000/262144, I/O @ 0x
00007ff8/8, BIOS @ 0x????????/131072
[    78.176] (--) PCI: (0:1:1:0) 8086:080d:1462:a380 rev 1, Mem @ 0xff010000/65536, I/O @ 0x00002100/32
[    78.176] (II) "glx" will be loaded by default.
[    78.176] (II) LoadModule: "dbe"
[    78.176] (II) Module "dbe" already built-in
[    78.176] (II) LoadModule: "glx"
[    78.181] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    78.344] (II) Module glx: vendor="X.Org Foundation"
[    78.345]    compiled for 1.19.6, module version = 1.0.0
[    78.345]    ABI class: X.Org Server Extension, version 10.0
[    78.345] (==) Matched modesetting as autoconfigured driver 0
[    78.345] (==) Matched fbdev as autoconfigured driver 1
[    78.345] (==) Matched vesa as autoconfigured driver 2
[    78.345] (==) Assigned the driver to the xf86ConfigLayout
[    78.345] (II) LoadModule: "modesetting"
[    78.345] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    78.414] (II) Module modesetting: vendor="X.Org Foundation"
[    78.414]    compiled for 1.19.6, module version = 1.19.6
[... 62 lines cut ...]
[    87.628] (II) UnloadSubModule: "fbdevhw"
[    87.628] (II) Unloading fbdevhw
[    87.628] (II) UnloadModule: "vesa"
[    87.628] (II) Unloading vesa
[    87.628] (==) Depth 24 pixmap format is 32 bpp
[    87.629] (==) modeset(0): Backing store enabled
[    87.629] (==) modeset(0): Silken mouse enabled
[    87.629] (II) modeset(0): RandR 1.2 enabled, ignore the following RandR disabled message.
[    87.629] (==) modeset(0): DPMS enabled
[    87.629] (--) RandR disabled
[    87.649] (II) SELinux: Disabled on system
[    87.652] (II) AIGLX: Screen 0 is not DRI2 capable
[    87.652] (EE) AIGLX: reverting to software rendering
[    87.670] (II) IGLX: enabled GLX_MESA_copy_sub_buffer
[    87.674] (II) IGLX: Loaded and initialized swrast
[    87.674] (II) GLX: Initialized DRISWRAST GL provider for screen 0
[    87.675] (II) modeset(0): Damage tracking initialized
[    87.676] (II) modeset(0): Setting screen physical size to 270 x 203
[    88.585] (II) config/udev: Adding input device Video Bus (/dev/input/event7)
[    88.585] (**) Video Bus: Applying InputClass "evdev keyboard catchall"
[    88.585] (**) Video Bus: Applying InputClass "libinput keyboard catchall"
[    88.585] (**) Video Bus: Applying InputClass "keyboard-all"
...

```

psergiu, 2022-12-19:
Please then update http://www.lakka.tv/doc/Hardware-support/

Section "**Generic PC**" from:

_Nvidia and Intel graphic cards should work. If you have to choose, choose Intel HD graphics for now._

to:

_Nvidia and Intel graphic cards (except GMA500) should work. If you have to choose, choose Intel HD graphics for now._

Thank you !

gouchi, 2022-12-19:
We will have to update the [wiki page](https://github.com/libretro/Lakka-LibreELEC/wiki/Hardware-support).