libretro/Lakka-LibreELEC #1797: HDMI Audio not working on Intel NUC 10

netthier, 2022-12-16:
- Which version of Lakka are you using?
`devel-20221119002255-5a6e401 (Generic.x86_64)`
- What system hardware are you using?
Intel NUC10i7FNH
- What did you do?
I attempted listening to my games
- What did you expect to happen?
For the games to produce sounds
- What happened instead?
They did not produce sounds.

`retroarch -v` outputs the following:
```
[ERROR] ALSA: Failed to initialize...
[ERROR] Failed to initialize audio driver. Will continue without audio.
```
`aplay -l` is empty, `aplay -L` lists only PCH devices.
The device does seem to be detected though:
```
# cat /proc/asound/card0/codec\#2
Codec: Intel Kabylake HDMI
```
Interestingly, `dmesg | grep "audio"` shows this:
```
[    3.633817] snd_hda_codec_hdmi hdaudioC0D2: HDMI: pin nid 5 not registered
[    3.751371] snd_hda_codec_hdmi hdaudioC0D2: HDMI: pin nid 5 not registered
```
which seems to be related to this kernel/ALSA bug patched in July 2021: https://github.com/clearlinux/distribution/issues/2396
I was unable to apply the workarounds mentioned in that thread as /lib/firmware is not writable and `hdajackretask` is a GUI tool.
Is it possible that this patch is not included in Lakka builds yet, or am I encountering a different issue?

gouchi, 2023-01-02:
Hi,

> I was unable to apply the workarounds mentioned in that thread as /lib/firmware is not writable

You can [add firmware](https://wiki.libreelec.tv/how-to/add-firmware) in `/storage/.config/firmware`.

If it still not working we might need to wait Lakka 5.x to make some test as we will be using newer kernel.

gouchi, 2023-01-25:
Can you try to make a test with [dev build from Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) ?

Thank you.

Zeron100Z, 2023-06-09:
> Can you try to make a test with [dev build from Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) ?
> 
> Thank you.

Hi @gouchi, sorry to revive a semi-old issue, but this issue still persists on Intel NUC Systems (or at least I can confirm it exists on a NUC6CAYH) on both the main and latest nightly build of Lakka

gouchi, 2023-06-09:
@Zeron100Z Please provide some [logs](https://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) otherwise it will be difficult to fix this issue. Thank you.

P.S = It might not be the same issue as you are not using Intel NUC 10.

gouchi, 2023-12-19:
@netthier Did you have the opportunity to make some test with [Lakka 5.x build](https://nightly.builds.lakka.tv/devbuild_latest/Generic.x86_64/) or [X11 build](https://nightly.builds.lakka.tv/devbuild_latest/x11.x86_64/) ?

Thank you.

ToKe79, 2024-04-04:
no response - feel free to re-open/submit new issue

closed 2024-04-04 by ToKe79