libretro/Lakka-LibreELEC #1798: Lakka USB stuck on flower logo when installing

CouncilBerr, 2022-12-17:
I am using the latest Lakka version  for x64 cpu for Windows 11 22H2 on a PC. PC has NTFS bios Intel Celeron N5095 CPU, 16GB DDR4 sodimm ram, 240GB SSD. I followed the steps to make a live boot USB 16GB USB 2.0 PNY brand. After installing Lakka and removing USB after restart, Lakka restarts with Lakka icon but stuck on this screen for hours. BIOS has all security features disabled and virtualization off.

gouchi, 2022-12-19:
Hi,

Can you try to provide some with [log](https://lakka.tv/doc/Troubleshooting-Lakka/#automated-collection-of-relevant-logs) with `createlog` ?

You can try to boot parameters with `live ssh textmode`  using your USB stick.

Thank you.

CouncilBerr, 2022-12-20:
Plugged in Lakka usb and did live ssh textmode and got a log file. Tested issue before it persists, hope the log file shows the issue. 
[log-2022-12-20-04.19.32.zip](https://github.com/libretro/Lakka-LibreELEC/files/10265317/log-2022-12-20-04.19.32.zip)

gouchi, 2022-12-21:
Hi,

Thank you for the log. 

Strange you are getting segfault using Intel GPU which is [enabled](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/projects/Generic/linux/linux.x86_64.conf#L4141) on Lakka.

May you try to make a test with [nightly build](https://nightly.builds.lakka.tv/latest/Generic.x86_64) ?

Thank you.

CouncilBerr, 2022-12-22:
> Hi,
>
> Thank you for the log.
>
> Strange you are getting segfault using Intel GPU which is [...]

I used the recommended nightly build on the same usb in live mode and got the same issue. Here is the log file. Thanks. 
[log-2022-12-22-00.26.54.zip](https://github.com/libretro/Lakka-LibreELEC/files/10282714/log-2022-12-22-00.26.54.zip)

barrystaes, 2022-12-24:
Similar problem. I have a fanless box with Intel igpu.

> CPU: Intel Jasper Lake Celeron Processor N5105, 4 core, 64 bit, 2.9GHz, 4M cache
> GPU: Intel UHD Grafische GPU, 24EU, 450MHz tot 800MHz

The log: [log-2022-12-24-21.50.26.zip](https://github.com/libretro/Lakka-LibreELEC/files/10299203/log-2022-12-24-21.50.26.zip)

> 01_RetroArch.log tail:
> ```
> [INFO] [Video]: Starting threaded video driver ...
> [ERROR] [KMS]: Couldn't find a suitable DRM device.
> [INFO] [GL]: Found GL context: "null".
> [INFO] [GL]: Detecting screen resolution: 320x240.
> [INFO] [GL]: Vendor: (null), Renderer: (null).
> [INFO] [GL]: Version: (null).
> Segmentation fault
> ```

barrystaes, 2022-12-24:
Tried again with **nightly build:** 
[log-2022-12-24-22.22.17.zip](https://github.com/libretro/Lakka-LibreELEC/files/10299217/log-2022-12-24-22.22.17.zip)

```
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: "null".
[INFO] [GL]: Detecting screen resolution: 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
[ERROR] [Video]: Cannot open threaded video driver ... Exiting ...
[ERROR] Fatal error received in: "video_driver_init_internal()"
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 0, Frames dropped: 0.
```

barrystaes, 2023-01-18:
@gouchi it seems this got marked as completed because the user who reported this has been deleted, and replaced by "ghost". 

Can you reopen this issue for others to gather more reproduction info on this problem?
If needed i have spare hardware to test nightly versions.

gouchi, 2023-01-18:
@barrystaes Done ;)

Can you [run](https://lakka.tv/doc/Accessing-Lakka-command-line-interface/) this command to list the available video outputs:

`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

You can try with [Lakka 4.3](https://le-builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-4.3.img.gz).

Thank you.

gouchi, 2023-12-19:
@barrystaes Any update on this issue, perhaps you did try using [Lakka 5.x](https://nightly.builds.lakka.tv/devbuild_latest/) ?

Thank you.

ToKe79, 2024-04-04:
no response - feel free to re-open with update / open new issue

closed 2024-04-04 by ToKe79