libretro/Lakka-LibreELEC #1799: Lakka build error "ERROR: libmp3lame >= 3.98.3 not found"

Jeremys556, 2022-12-18:
Hello,
I downloaded the source code for Lakka and I am trying to build it. I ran the command on the docs that is:
`DISTRO=Lakka PROJECT=Generic ARCH=x86_64 make image`

However, I am running into the following error:
```ERROR: libmp3lame >= 3.98.3 not found

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.libera.chat.
Include the log file "ffbuild/config.log" produced by configure as this will help
solve the problem.
FAILURE: scripts/build ffmpeg:target during configure_target (package.mk)
*********** FAILED COMMAND ***********
./configure --prefix="/usr" --cpu="${TARGET_CPU}" --arch="${TARGET_ARCH}" --enable-cross-compile --cross-prefix="${TARGET_PREFIX}" --sysroot="${SYSROOT_PREFIX}" --sysinclude="${SYSROOT_PREFIX}/usr/include" --target-os="linux" --nm="${NM}" --ar="${AR}" --as="${CC}" --cc="${CC}" --ld="${CC}" --host-cc="${HOST_CC}" --host-cflags="${HOST_CFLAGS}" --host-ldflags="${HOST_LDFLAGS}" --extra-cflags="-I/usr/local/src/lame-3.100" --extra-ldflags="-L/usr/local/src/lame-3.100" --pkg-config="${TOOLCHAIN}/bin/pkg-config" ${CONFIG_OPTIONS_STANDARD_FFMPEG}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install ffmpeg:target has failed!

The following log for this failure is available:
  /home/jeremys556/Lakka-LibreELEC/build.Lakka-Generic.x86_64/.threads/logs/198.log

>>> ffmpeg:target seq 198 >>>
[193/387] [FAIL] install ffmpeg:target

The following log for this failure is available:
  /home/jeremys556/Lakka-LibreELEC/build.Lakka-Generic.x86_64/.threads/logs/198.log

Parallel build failure - see log for details. Time of failure: Sat Dec 17 21:38:24 PST 2022
make: *** [Makefile:10: image] Error 1
```

I attempted to manually install LAME, which worked, but this still does not work.
I also found the following stackoverflow forum, but even after following directions on the forum, it still does not appear to work: https://stackoverflow.com/questions/35937403/error-libmp3lame-3-98-3-not-found

I am running this on Ubuntu 20 LTS.

Can anyone help? Been stuck on this for like two hours...

gouchi, 2022-12-19:
Hi,

Can you try to add [this condition for FFmpeg package](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-v4.x/packages/multimedia/ffmpeg/package.mk#L12-L14).

We are still working on stabilizing Lakka 5.x branch.

Please let us know how it goes.

Thank you.

victor-oliveira1, 2022-12-22:
I can compile ffmpeg using the condition suggested.
Thank you!

gouchi, 2022-12-22:
Thank you for the feedback.

Kind regards,

closed 2022-12-22 by gouchi