libretro/Lakka-LibreELEC #1806: Wifi, bluetooth and USB Ports not working on Rock Pi 4C+

maknaibs, 2022-12-29:
On Rock Pi 4C+ models with the latest nightly image installed from [here ](https://nightly.builds.lakka.tv/latest/RK3399.aarch64/) (Lakka-RK3399.aarch64-nightly-v4.x-20221226-591fe24-rock-pi-4.img.gz), the following things are not working:

- only 1 out of 4 USB ports are recognized
- No Wifi networks can be found, ethernet is fine
- Bluetooth controllers can't be paired
- there are problems with selecting any resolution other than 4k

Jelte2357, 2023-02-18:
i have the same problems exept for the 1/4 usb on the pine64 A64

paqman3d, 2023-06-22:
Same issue with my Rock 4C+. It's been a few months since I last tried a nightly, but even Kodi has a rough time with this board -- so this seems to be a crappy board despite the specs lol. 

In Lakka, it seems to be outputting in 4K by default as well, which slows down the XMB to a slide show. I don't think there's a way to change the resolution to something more sensible like 720p, either. If I could change the res, that would go a long way with performance. I can't even get past the PS1 bios lol.

The other USB ports, wifi and Bluetooth not working are a bigger problem, though. If I slap Linux or Android on the board, most of the features work. It's not a total loss, but hopefully LibreELEC can iron these things out.

jptuomi, 2024-04-02:
I had success (not with Lakka though) in using the images from here:
https://forum.libreelec.tv/thread/20823-unofficial-le-for-rk356x-rk3328-rk3399-rk3588-s/?pageNo=1

But what the instructions fail to mention clearly enough IMO is that you need to edit the /extlinux/extlinux.conf file to correspond with the .dtb-file of your SBC.

paqman3d, 2024-06-21:
As of 6.X, wifi works.. sometimes. It will connect, then forget credentials and just not work until a reboot. I tried editing the wifi-config file with my network information, but it still is spotty. 

Bluetooth can see devices but I can't pair. Bluetoothctl says "no default controller available" when attempting to pair. Using a external adapter works in the meantime.

paqman3d, 2026-09-15:
> [@maknaibs](https://github.com/maknaibs) [@jptuomi](https://github.com/jptuomi) [@paqman3d](https://github.com/paqman3d) Is it still an issue using [latest nightly](https://nightly.builds.lakka.tv/latest/RK3399.aarch64/) ?
> 
> Thank you.

I'll get to testing it and report back shortly.