libretro/Lakka-LibreELEC #1810: Black screen on Mupen64plus_next running with Amlogic generic image

victor-oliveira1, 2022-12-30:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka-AMLGX.aarch64-nightly-v4.x-20221226-591fe24-box.img.gz
- What system hardware are you using?
Amlogic AMLGX s905x3
- What did you do?
Installed the Lakka 4.2 and nightly image
- What did you expect to happen?
Played some Nintendo 64 game using mupen64plus_next core
- What happened instead?
The games showed a black screen with audio running on background

I've fixed the black screen problem adding the FORCE_GLES3=1 on mupen64plus_next package.mk file

h0rseduck, 2023-01-09:
I have this problem with Nintendo 64 (mupen64plus_next) on Orange Pi 3 LTS. Black screen with audio running on background. The menu opens and works well.

Version Lakka-H6.aarch64-4.2-orangepi-3

FORCE_GLES3=1 in mupen64plus_next/package.mk doesn't help

```
[INFO] [Environ]: SET_SUBSYSTEM_INFO.
[INFO] [Core]: Using content: "/storage/roms/n64/Worms - Armageddon (U) (M3) [!].z64".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 1185, Frames dropped: 0.
[INFO] RetroArch 1.10.3 (Git 9b282aa)
[INFO] === Build =======================================
[INFO] Возможности:  ASIMD
[INFO] Built: Apr 24 2022
[INFO] Version: 1.10.3
[INFO] Git: 9b282aa
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/mupen64plus_next_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/Mupen64Plus-Next.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/n64.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/storage/.config/retroarch/config/Mupen64Plus-Next/Worms - Armageddon (U) (M3) [!].cfg".
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Remaps]: Remap directory: "/storage/remappings".
[INFO] [Overrides]: Файл карты памяти перенаправлен в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [Overrides]: Файл сохранения перенаправлен в "/storage/savestates/Worms - Armageddon (U) (M3) [!].state".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Content]: Загрузка контента: "/storage/roms/n64/Worms - Armageddon (U) (M3) [!].z64".
[INFO] [Content]: Не найден правильный патч для контента.
[INFO] [Environ]: SET_HW_RENDER, context type: gl.
[INFO] Requesting OpenGLES2 context.
[INFO] Reached end of SET_HW_RENDER.
[libretro INFO] mupen64plus: Using full mem base
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Goodname: Worms - Armageddon (U) (M3) [!]
[libretro INFO] mupen64plus: Name: WORMS ARMAGEDDON    
[libretro INFO] mupen64plus: MD5: 491D4DB6718302489BF05FB962C73651
[libretro INFO] mupen64plus: CRC: 13E959A0 0E93CAB0
[libretro INFO] mupen64plus: Imagetype: .z64 (native)
[libretro INFO] mupen64plus: Rom size: 12582912 bytes (or 12 Mb or 96 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 1444
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 4441
[libretro INFO] mupen64plus: Country: USA
[libretro INFO] mupen64plus: PC = 80000400
[libretro INFO] mupen64plus: Save type: 0
[INFO] [Core]: Версия API libretro: 1, Скомпилирован против API: 1
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Set video size to: 1440x1080.
[INFO] [Video]: Using HW render, OpenGL driver forced.
[INFO] [Video]: Using configured "gl" driver for GL HW render.
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 29 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 30.000000 Hz
[... 104 lines cut ...]
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X102
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Initializing 4 RDRAM modules for a total of 8 MB
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [Config]: Новая конфигурация сохранена в "/storage/.config/retroarch/retroarch.cfg".
[INFO] [SRAM]: Запись RAM #0 в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [SRAM]: Успешно сохранено в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 19 seconds.
[INFO] [Runtime]: Saving runtime log file: "/storage/playlists/logs/Mupen64Plus-Next/Worms - Armageddon (U) (M3) [!].lrtl".
[INFO] [Core]: Unloading game..
[INFO] [Core]: Unloading core..
[libretro INFO] mupen64plus: Stopping emulation.
[libretro INFO] mupen64plus: R4300 emulator finished.
[INFO] [PERF]: Performance counters (libretro):
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved core options file to "/storage/.config/retroarch/config/Mupen64Plus-Next/Mupen64Plus-Next.opt".
```

victor-oliveira1, 2023-01-11:
> I have this problem with Nintendo 64 (mupen64plus_next) on Orange Pi 3 LTS. Black screen with audio running on background. The menu opens and works well.
>
> Version Lakka-H6.aarch64-4.2-orangepi-3 [...]

Hi!

I think there's something wrong with your build flags. 
Look at the line:
`[INFO] Requesting OpenGLES2 context. `
The emulator it's requesting OpenGLES2, not OpenGLES3 context. 
Here's what iv'e changed in the **packages/lakka/libretro_cores/mupen64plus_next/package.mk** file:  
```
 if [ "${OPENGLES_SUPPORT}" = "yes" ]; then
  PKG_DEPENDS_TARGET+=" ${OPENGLES}"
  PKG_MAKE_OPTS_TARGET+=" GLES=1 FORCE_GLES=1"
fi
```

Thanks!

Ntemis, 2023-01-12:
H6 only supports GLES2 on panfrost

victor-oliveira1, 2023-01-12:
> H6 only supports GLES2 on panfrost

@Ntemis, sad but true, the logs shows that too:
> [INFO] [GL]: Version: OpenGL ES 2.0 Mesa 22.0.2.

@h0rseduck, it's not possible to use GLES3 context on this device.

:(

h0rseduck, 2023-01-12:
I don't understand how can I run N64 with mupen64plus_next emulator. I have a black screen with audio running on background. The menu opens and works well.

The Parellel-N64 emulator starts fine but has very low FPS. I tried different core settings but they don't help.

Ntemis, 2023-01-20:
Guys please test lakka 4.3 and let me know, it should be fixed.

h0rseduck, 2023-01-21:
Nothing has changed for me. 
Lakka-H6.aarch64-4.3-orangepi-3

```
[INFO] [Environ]: SET_SUBSYSTEM_INFO.
[INFO] [Core]: Using content: "/storage/roms/n64/Worms - Armageddon (U) (M3) [!].z64".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 3687, Frames dropped: 0.
[INFO] RetroArch 1.14.0 (Git ad89b0c)
[INFO] === Build =======================================
[INFO] Возможности: ASIMD
[INFO] Built: Jan 16 2023
[INFO] Version: 1.14.0
[INFO] Git: ad89b0c
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/tmp/cores/mupen64plus_next_libretro.so"
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_V2.
[INFO] [Environ]: SET_CONTROLLER_INFO.
[INFO] [Overrides]: Файл карты памяти перенаправлен в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [Overrides]: Файл сохранения перенаправлен в "/storage/savestates/Worms - Armageddon (U) (M3) [!].state".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_PERF_INTERFACE.
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Content]: Загрузка контента: "/storage/roms/n64/Worms - Armageddon (U) (M3) [!].z64".
[INFO] [Environ]: SET_HW_RENDER, context type: gl.
[INFO] Requesting OpenGLES2 context.
[INFO] Reached end of SET_HW_RENDER.
[libretro INFO] mupen64plus: Using full mem base
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Goodname: Worms - Armageddon (U) (M3) [!]
[libretro INFO] mupen64plus: Name: WORMS ARMAGEDDON    
[libretro INFO] mupen64plus: MD5: 491D4DB6718302489BF05FB962C73651
[libretro INFO] mupen64plus: CRC: 13E959A0 0E93CAB0
[libretro INFO] mupen64plus: Imagetype: .z64 (native)
[libretro INFO] mupen64plus: Rom size: 12582912 bytes (or 12 Mb or 96 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 1444
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 4441
[libretro INFO] mupen64plus: Country: USA
[libretro INFO] mupen64plus: PC = 80000400
[libretro INFO] mupen64plus: Save type: 0
[INFO] [Core]: Версия API libretro: 1, Скомпилирован против API: 1
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Set video size to: 1440x1080.
[INFO] [Video]: Using HW render, OpenGL driver forced.
[INFO] [Video]: Using configured "gl" driver for GL HW render.
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 27 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 2: (1680x1050) 1680 x 1050, 59.883251 Hz
[INFO] [DRM]: Mode 3: (1600x900) 1600 x 900, 60.000000 Hz
[INFO] [DRM]: Mode 4: (1280x1024) 1280 x 1024, 75.024673 Hz
[INFO] [DRM]: Mode 5: (1280x1024) 1280 x 1024, 60.019741 Hz
[INFO] [DRM]: Mode 6: (1440x900) 1440 x 900, 59.901459 Hz
[INFO] [DRM]: Mode 7: (1280x800) 1280 x 800, 59.909546 Hz
[INFO] [DRM]: Mode 8: (1152x864) 1152 x 864, 75.000000 Hz
[INFO] [DRM]: Mode 9: (1280x720) 1280 x 720, 60.000000 Hz
[... 97 lines cut ...]
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X102
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[INFO] [Environ]: SYSTEM_DIRECTORY: "/tmp/system".
[libretro INFO] mupen64plus: Initializing 4 RDRAM modules for a total of 8 MB
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [Config]: Новая конфигурация сохранена в "/storage/.config/retroarch/retroarch.cfg".
[INFO] [SRAM]: Запись RAM #0 в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [SRAM]: Успешно сохранено в "/storage/savefiles/Worms - Armageddon (U) (M3) [!].srm".
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 14 seconds.
[INFO] [Runtime]: Saving runtime log file: "/storage/playlists/logs/Mupen64Plus-Next/Worms - Armageddon (U) (M3) [!].lrtl".
[INFO] [Core]: Unloading game..
[INFO] [Core]: Unloading core..
[libretro INFO] mupen64plus: Stopping emulation.
[libretro INFO] mupen64plus: R4300 emulator finished.
[INFO] [PERF]: Performance counters (libretro):
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved core options file to "/storage/.config/retroarch/config/Mupen64Plus-Next/Mupen64Plus-Next.opt".
```

T-SUGIUE, 2023-03-31:
I am also facing the same issue.

-What system hardware are you using?
Shenzhen Xunlong Software Orange Pi PC2

-Which version of Lakka are you using? 
Lakka-H5.aarch64-4.3-orangepi-pc2.img
Lakka-H5.aarch64-nightly-v4.x-20230211-06af145-orangepi-pc2.img
Lakka-H5.aarch64-devbuild-v5.x-20230226-125aa20-orangepi-pc2.img

What kind of logs would help?

orangechickenpasta, 2023-04-19:
I am experiencing the same issue using Lakka-H6.aarch64-nightly-v4.x-20230411-948d92b-orangepi-3.img

slowpokefarm, 2023-04-22:
Same issue on Lakka-H6.aarch64-devbuild-v5.x-20230226-125aa20-orangepi-3-lts image. SOME roms do show image though.

Ntemis, 2023-04-22:
Allwinner H6 does not support OpenGL ES v3

Ntemis, 2023-04-22:
The initial issue was reported on Amlogic AMLGX s905x3 that SUPPORTS  OpenGL ES v3 and that now that issue is fixed.
Closing down

slowpokefarm, 2023-04-22:
Ah, ok. I thought lakka for H6 is shipped with GLES2 version of Mupen64plus_next like on Android. For anyone coming here from google search - you can still use Mupen64plus_next on H6 if you are running Android with RetroArch instead of Lakka.

Ntemis, 2023-04-22:
@slowpokefarm we can cheat the build system, please try next nightly
;)

closed 2023-04-22 by Ntemis