libretro/Lakka-LibreELEC #1813: Dualshock 4 Still not working?

XtremeOwnageDotCom, 2023-01-01:
A few years back, I built a HTPC for my livingroom, and decided to use Lakka. I ended up picking up a pair of dual shock 4 controllers to use with it.

Back then, I had nothing but issues trying to get the controllers to work properly... and I ended up buying a pair of xbox one controllers, which worked much better.

Today, I decided to attempt to build a retrogaming setup for my kids, so they can experience the joy of couch coop, and learn to curse while playing mario bros.

So, I downloaded the latest version of Lakka `Generic.x86_64-4.2`, and installed this to an SSD on a Optiplex 3060.

This OptiPlex 3060, has a [Intel Wireless-AC 9560](https://www.intel.com/content/www/us/en/products/sku/99446/intel-wirelessac-9560/specifications.html) which does have Bluetooth support.

After installing Lakka. I goto the Bluetooth settings page, and pair my dual shock 4 controllers. 

They then show as connected. One of the controllers lights up the LED as Blue.

However, at this point, none of the buttons on the controller will do anything. Going to the Input 1 -> Map Device, I can see, "Wireless controller" is in the slot. 

But, none of the buttons will do anything.

Am, I missing a key piece here?

My expectation, install lakka, on somewhat modern hardware, using well-supported intel hardware/bluetooth, plug in a well-known controller, and everything should work.. But, my expectations have fallen quite short of reality.

In the system information page, I CAN see:

Port 0 device name: Wireless Controller (#0)
Device config name: sony-playstation4-dualshock4v2-controller.cfg
device identifiers: 1356/2508

So, it would appear, the controller is properly detected.

Another oddity,

If I visit the bluetooth discovery page, while the controller is on, it will somewhat start working.

XtremeOwnageDotCom, 2023-01-01:
After trying lots of things, and plugging in a cheap tp-link Bluetooth adaptor, and disabling the on-board Bluetooth adapter in the bios- 

Everything paired up and worked the first try, and seems to work correctly.

I would say, there may be a compatibility issue with the Intel 9560 adaptor.

gouchi, 2023-01-02:
You might try to provide [log](https://www.lakka.tv/doc/Troubleshooting-Lakka/) with `createlog` and also `journalctl -u bluetooth`.

Also did you make a test with [nightly build](https://nightly.builds.lakka.tv/latest/Generic.x86_64/) ?

Thank you.

XtremeOwnageDotCom, 2023-01-02:
Removed tp-link adaptor, re-enabled intel 9560 Bluetooth adaptor in BIOS.

After starting, both controllers actually paired and worked without issue. 

*** Lots of testing / diagnosis steps omitted here.

I think the resolution to my issue was screwing on an antenna... After attaching an antenna, everything seems to work normally.

My apologizes for producing an PEBKAC issue for you.

closed 2023-01-02 by XtremeOwnageDotCom