libretro/Lakka-LibreELEC #1817: Support for Rockchip RK3588S/RK3588?

LAP87, 2023-01-09:
Orange Pi 5 is out and starting to arrive at early buyers,
Any plans to add this to the great list of hardware that can run Lakka?

/Best regards

xtc0r, 2023-01-23:
I own two [Rock 5B](https://wiki.radxa.com/Rock5/5B) which have the RK3588 as well. It would be a powerful gaming machine.

port19x, 2023-05-26:
There are several attractive products with rk3588s and rk3588 chips out now.
I'd love to see support

Poincare1, 2024-06-15:
More than one year later. Any hope for Lakka as booting and playable on RK3588 boards such as Orange Pi 5b??

mannes, 2025-01-14:
https://github.com/LibreELEC/LibreELEC.tv/pull/7864 there's work being done upstream for support of RK3588 and RK356x. Once that PR get's through it'll make it's way to Lakka I expect.

Right now you could help validating the work in LibreELEC to perhaps speed up the process

ShigeakiAsai, 2025-08-25:
https://github.com/LibreELEC/LibreELEC.tv/pull/10420
Rockchip: add support for RK3588, RK3576, and RK356X devices

upstream LibreELEC re-starts support Rockchip devices, I'm watching this.

atirage21, 2026-09-30:
Under retroarch at rk3588 libreeelec 13 are not configuration files or something, because are only fake downloaded kernels without real saved at system. Is not possible to use RA at rk3588 without kernels. https://forum.libreelec.tv/thread/29500-libreelec-lakka-retroarch/?postID=207586#post207586

ShigeakiAsai, 2026-09-30:
I'm not a member of the Lakka team, so this is just my personal opinion.

I don't think either the LibreELEC or the Lakka team would support running Lakka's RetroArch on LibreELEC.

- The Flatpak RetroArch problem on LE13 should be reported on the LibreELEC forum rather than here. LibreELEC generally prefers forum reports over GitHub issues.
- As seen in the forum thread, the Lakka 6.1 RetroArch build fails on LibreELEC because libSDL2 is missing. Newer Lakka nightlies have also moved further toward SDL3, so this is unlikely to get easier.
- Lakka's devel branch, which includes RK35xx support, is currently not being updated, and v6.1 has no RK356X/RK3576/RK3588 support. There is also no RK3588 build on the Lakka v6.1 nightly site, so the script has nothing to download for that device. Official RK3588 support in Lakka is probably still some way off.

So for now, please report the issue on the LibreELEC side and wait for a fix there.