libretro/Lakka-LibreELEC #1824: Lakka and Stadia Controller (Bluetooth Mode) does not work together

mtoensing, 2023-01-20:
Please make Stadia Controllers compatible with Lakka. 

- Which version of Lakka are you using?
- Latest release version
- What system hardware are you using?
- RP4
- What did you do?
- Try to pair Stadia Controller and use it in Lakka
- What did you expect to happen?
- Automatic configuration of the Stadia Controller
- What happened instead?
- Nothing. Did not work. Xbox 360 controllers with USB dongle work fine

Ntemis, 2023-01-20:
I dont have a stadia controller unfortunately
You can send the config here and will be added to the next lakka version 
https://github.com/libretro/retroarch-joypad-autoconfig

mtoensing, 2023-01-20:
Hi, what do you mean mit send? I can send you controller if you like.

Ntemis, 2023-01-20:
https://discord.gg/libretro-184109094070779904
come to #lakkatv and pm me same nick as here

gouchi, 2023-02-04:
It seems  you need to update your controller to use Bluetooth ?

Source [1](https://www.gamingonlinux.com/2023/01/bluetooth-support-for-the-stadia-controller-is-now-live/) [2](https://support.google.com/stadia/answer/13067284)

mtoensing, 2023-02-04:
@gouchi 
My Stadia controller is updated to use Bluetooth and works fine on macOS and Steam Deck without any configuration. I will send @Ntemis a controller if I get hold of enough controllers to have a spare one. Since he can't just buy one where he lives this is still on my agenda since donating money is not accepted here.

ToKe79, 2023-02-04:
@mtoensing were you able to pair the controller via bluetooth? Maybe pairing via RetroArch UI will not work, so please try using command line.
[Guide - SSH access](https://github.com/libretro/Lakka-LibreELEC/wiki/Accessing-Lakka-command-line-interface#ssh)
[Guide - bluetoothctl](https://github.com/libretro/Lakka-LibreELEC/wiki/8Bitdo-Wireless-Controller)

After you pair the controller, you might need to manually map controls to actions.

mtoensing, 2023-02-04:
I know I could do that. But since there are now millions of Stadia controllers in the wild I would appreciate that this works automatically in Lakka.

ToKe79, 2023-02-04:
That is the reason why I kindly ask you to pair it first - to know if there is support in the linux kernel already (meaning only the joypad mapping is necessary) or a kernel driver/module needs to be added.

marevalo, 2023-02-07:
I got it to connect correctly:

```
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:6A:B6:16 Discovering: yes
[NEW] Device 54:F8:2A:52:B7:33 54-F8-2A-52-B7-33
[NEW] Device 4E:78:8D:7C:DF:11 4E-78-8D-7C-DF-11
[NEW] Device 00:C3:F4:E8:28:5D 00-C3-F4-E8-28-5D
[NEW] Device F0:EB:0F:53:A5:4C LH719
[NEW] Device 60:35:73:E6:F0:2E MiTV-AESP0 5151
[NEW] Device E4:CF:33:D9:C6:94 StadiaHFW8-c694
[CHG] Device 00:C3:F4:E8:28:5D ManufacturerData Key: 0x0075
[CHG] Device 00:C3:F4:E8:28:5D ManufacturerData Value:
  42 04 01 20 76 19 0f 00 00 01 32 00 00 00 00 00  B.. v.....2.....
  00 00 00 00 00 00 00 00                          ........        
[bluetooth]# connect E4:CF:33:D9:C6:94
Attempting to connect to E4:CF:33:D9:C6:94
[CHG] Device E4:CF:33:D9:C6:94 Connected: yes
Connection successful
[NEW] Primary Service (Handle 0x4b20)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0001
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0001/char0002
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor (Handle 0x001c)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0001/char0002/desc0004
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0x60a8)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028/char0029
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028/char002b
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028/char002d
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028/char002f
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0028/char0031
	00002a50-0000-1000-8000-00805f9b34fb
	PnP ID
[NEW] Primary Service (Handle 0x80f0)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0038
	0000180f-0000-1000-8000-00805f9b34fb
	Battery Service
[NEW] Characteristic (Handle 0x0015)
	/org/bluez/hci0/dev_E4_CF_33_D9_C6_94/service0038/char0039
	00002a19-0000-1000-8000-00805f9b34fb
	Battery Level
[CHG] Device E4:CF:33:D9:C6:94 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device E4:CF:33:D9:C6:94 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device E4:CF:33:D9:C6:94 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device E4:CF:33:D9:C6:94 UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[CHG] Device E4:CF:33:D9:C6:94 UUIDs: 00001812-0000-1000-8000-00805f9b34fb
[CHG] Device E4:CF:33:D9:C6:94 ServicesResolved: yes
[CHG] Device E4:CF:33:D9:C6:94 WakeAllowed: yes
[CHG] Device E4:CF:33:D9:C6:94 Modalias: usb:v18D1p9400d0100
```
After pairing this is the info I get:

```[StadiaHFW8-c694]# info
Device E4:CF:33:D9:C6:94 (public)
	Name: StadiaHFW8-c694
	Alias: StadiaHFW8-c694
	Appearance: 0x03c4
	Icon: input-gaming
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v18D1p9400d0100
	RSSI: -40
	TxPower: -8
```
After that the controller sometimes appears on the bluetooth screen of RetroArch with a generic bluetooth icon, but it is not available on the input mapping screen options.

I'm running lakka stable (4.3) on raspberry pi 3.

On my Debian Stable box the controller is connected and setup with the graphical standard process without problem (well, mostly, some random disconnections).

ToKe79, 2023-02-07:
Thanks, if the device is recognized correctly as gamepad/joystick, you should be able to test the inputs via command line, e.g. `jstest /dev/input/js0`. If not, you can still try to map actions to buttons/sticks in RetroArch (use other gamepad / keyboard to navigate the menus).

Now that it is paired with your RPi, it *should* connect after reboot without the need of the command line. Do you get a brief message that a gamepad was detected in RetroArch? For example something in the meaning that no valid configuration was found? It is possible that it is not recognized as gamepad, but other type HID device (keyboard/mouse/...) - in such case there will be no such message. Also can you paste kernel log (`dmesg`) with the messages when the gamepad gets connected?

marevalo, 2023-02-07:
Hello.

I don't have any /dev/input/js0 (custom udev rules needed?). And so I cannot map actions as there is no device index available.

Yes, it reconnects (but with Paired: no); and no there is no message at the start or when switching off and on again the controller.

dmesg is rather cryptic:

```
[   19.636977] Bluetooth: Core ver 2.22
[   19.637084] NET: Registered protocol family 31
[   19.637091] Bluetooth: HCI device and connection manager initialized
[   19.637114] Bluetooth: HCI socket layer initialized
[   19.637126] Bluetooth: L2CAP socket layer initialized
[   19.637146] Bluetooth: SCO socket layer initialized
[   19.665205] Bluetooth: HCI UART driver ver 2.3
[   19.665220] Bluetooth: HCI UART protocol H4 registered
[   19.665301] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   19.665502] Bluetooth: HCI UART protocol Broadcom registered
[   20.548129] cryptd: max_cpu_qlen set to 1000
[   21.039794] Bluetooth: hci0: security requested but not available
[   33.758452] cam-dummy-reg: disabling
[   36.705382] Bluetooth: hci0: security requested but not available
[   70.491694] Bluetooth: hci0: security requested but not available
[  104.491927] Bluetooth: hci0: security requested but not available
[  110.965676] Bluetooth: hci0: security requested but not available
[  138.495823] Bluetooth: hci0: security requested but not available
```
BTMon output at controller switch on and connect:

```
Lakka:/var/log # btmon
Bluetooth monitor ver 5.56
= Note: Linux version 5.10.110 (armv7l)                                                                           0.368863
= Note: Bluetooth subsystem version 2.22                                                                          0.368877
= New Index: B8:27:EB:6A:B6:16 (Primary,UART,hci0)                                                         [hci0] 0.368881
= Open Index: B8:27:EB:6A:B6:16                                                                            [hci0] 0.368884
= Index Info: B8:27:EB:6A:B6:16 (Broadcom Corporation)                                                     [hci0] 0.368891
@ MGMT Open: bluetoothd (privileged) version 1.18                                                        {0x0001} 0.368896
> HCI Event: LE Meta Event (0x3e) plen 12                                                              #1 [hci0] 10.408903
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: E4:CF:33:D9:C6:94 (OUI E4-CF-33)
        Data length: 0
        RSSI: -41 dBm (0xd7)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                 #2 [hci0] 10.409047
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                            #3 [hci0] 10.410990
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection (0x08|0x000d) plen 25                                              #4 [hci0] 10.411093
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Public (0x00)
        Peer address: E4:CF:33:D9:C6:94 (OUI E4-CF-33)
        Own address type: Public (0x00)
        Min connection interval: 7.50 msec (0x0006)
        Max connection interval: 10.00 msec (0x0008)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                              #5 [hci0] 10.412004
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                                                              #6 [hci0] 10.439165
[... 128 lines cut ...]
        Handle: 0x001b
        Value: 021c00a62a
< ACL Data TX: Handle 64 flags 0x00 dlen 11                                                           #33 [hci0] 10.703455
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x001c-0x001c
        Attribute type: Characteristic (0x2803)
> ACL Data RX: Handle 64 flags 0x02 dlen 9                                                            #34 [hci0] 10.731660
      ATT: Error Response (0x01) len 4
        Read By Type Request (0x08)
        Handle: 0x001c
        Error: Attribute Not Found (0x0a)
< ACL Data TX: Handle 64 flags 0x00 dlen 11                                                           #35 [hci0] 10.732378
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x0000-0x0027
        Attribute type: Include (0x2802)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                #36 [hci0] 10.739211
        Num handles: 1
        Handle: 64
        Count: 2
> ACL Data RX: Handle 64 flags 0x02 dlen 9                                                            #37 [hci0] 10.746669
      ATT: Error Response (0x01) len 4
        Read By Type Request (0x08)
        Handle: 0x0000
        Error: Invalid Handle (0x01)
```

Fainought, 2023-02-17:
Hi, I have the same problem and came here to ask for the same thing. I can help too.

I noticed from this thread some errors or at least things that don't apply to my machine.

The stadia bluetooth enabled controller does connect to lakka via bluetooth both via ssh and gui.
The real problem is that it doesn't show up as a controller at all, no pop up and no controller in list.

Plus, the stadia controller works flawlessly on retroarch for windows. It even shows up the correct controller's name.

gouchi, 2023-02-24:
Is it also an issue using [dev builds](https://nightly.builds.lakka.tv/devbuild_latest/) from Lakka 5.x branch ?

Thank you.

Fainought, 2023-02-25:
No, it doesn't work even there. Same problem: gets connected but not recognized as a controller

steigerdich, 2023-06-21:
Do you still have that problem? I figured out, that it helps to restart bluetooth with a startup script. i added sleep 30, otherwise it won't work

gouchi, 2023-12-19:
Hi,

Is it still not working using [latest dev builds](https://nightly.builds.lakka.tv/devbuild_latest/) ?

Thank you.

nilo85, 2024-01-28:
@gouchi I have the same issue, I just tried installing the latest dev build (Lakka-RPi4.aarch64-devbuild-v5.x-20240125-9297f90.img.gz) and same issue there.

Anything I can do to assist or provide more details?

nilo85, 2024-01-28:
I stumbled upon this https://codeberg.org/fabiscafe/game-devices-udev
Gave it a try but not knowing much of this ecosystem of sofware, I suspect it requires a kernel module called uinput, which is not available when i lsmod..

gouchi, 2024-02-02:
@nilo85 Can you try to copy the [file](https://codeberg.org/fabiscafe/game-devices-udev/raw/branch/main/71-google-controllers.rules) in `/storage/.config/udev.rules.d/` ?

Then reload udev rules `udevadm control --reload-rules && udevadm trigger`

nilo85, 2024-02-02:
Still did not work =/ When u run the control reload and trigger UI showed that my PS3 controller disconnected (but still works) so that seems to have done something.

Is there a command to debug what udev sees?

nilo85, 2024-02-02:
when enabling debug output as in https://superuser.com/questions/677106/how-can-i-check-if-a-udev-rule-fired (top response)

I get :
`Feb 02 23:20:54 Lakka systemd-udevd[378]: Cleanup idle workers
Feb 02 23:20:56 Lakka systemd-udevd[378]: hci0:64: Device is queued (SEQNUM=1931, ACTION=add)
Feb 02 23:20:56 Lakka systemd-udevd[378]: hci0:64: Device ready for processing (SEQNUM=1931, ACTION=add)
Feb 02 23:20:56 Lakka systemd-udevd[378]: Successfully forked off '(udev-worker)' as PID 931.
Feb 02 23:20:56 Lakka systemd-udevd[378]: hci0:64: Worker [931] is forked for processing SEQNUM=1931.
Feb 02 23:20:56 Lakka (udev-worker)[931]: hci0:64: Processing device (SEQNUM=1931, ACTION=add)
Feb 02 23:20:56 Lakka (udev-worker)[931]: hci0:64: sd-device: Created db file '/run/udev/data/+bluetooth:hci0:64' for '/devices/platform/soc/fe201000.serial/serial0/serial0-0/bluetooth/hci0/hci0:64'
Feb 02 23:20:56 Lakka (udev-worker)[931]: hci0:64: Device processed (SEQNUM=1931, ACTION=add)
Feb 02 23:20:56 Lakka (udev-worker)[931]: hci0:64: sd-device-monitor(worker): Passed 394 byte to netlink monitor.
Feb 02 23:20:56 Lakka systemd-udevd[378]: No events are queued, removing /run/udev/queue.
Feb 02 23:20:56 Lakka bluetoothd[869]: src/service.c:service_accept() input-hog profile accept failed for F6:8B:AB:35:A8:F3
Feb 02 23:20:56 Lakka kernel: Bluetooth: hci0: security requested but not available
Feb 02 23:20:59 Lakka systemd-udevd[378]: Cleanup idle workers
Feb 02 23:20:59 Lakka (udev-worker)[931]: Unload kernel module index.
Feb 02 23:20:59 Lakka (udev-worker)[931]: Unloaded link configuration context.
Feb 02 23:20:59 Lakka systemd-udevd[378]: Worker [931] exited.`

gouchi, 2024-02-02:
> Is there a command to debug what udev sees?

you could try something like `udevadm monitor --property --udev`

nilo85, 2024-02-03:
This is what I get when booting the device (already paired) and when turning it off..

But from previous logs and some googling on simiar errors sonds like there might be an issue with bluez?
"Feb 02 23:20:56 Lakka bluetoothd[869]: src/service.c:service_accept() input-hog profile accept failed for F6:8B:AB:35:A8:F3" 

It seems to me the device is connecting and disconnecting properly but something fails on my machine during this and its only partially setup?

```
UDEV  [3292.827558] add      /devices/platform/soc/fe201000.serial/serial0/serial0-0/bluetooth/hci0/hci0:64 (bluetooth)
ACTION=add
DEVPATH=/devices/platform/soc/fe201000.serial/serial0/serial0-0/bluetooth/hci0/hci0:64
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=1977
USEC_INITIALIZED=3292826900
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:64
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:

UDEV  [3314.331092] remove   /devices/platform/soc/fe201000.serial/serial0/serial0-0/bluetooth/hci0/hci0:64 (bluetooth)
ACTION=remove
DEVPATH=/devices/platform/soc/fe201000.serial/serial0/serial0-0/bluetooth/hci0/hci0:64
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=1978
USEC_INITIALIZED=3292826900
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:64
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:
```

nilo85, 2024-02-03:
Also I see all these 
```
[ 3999.447873] Bluetooth: hci0: security requested but not available
```

https://www.linuxquestions.org/questions/slackware-14/logitech-bluetooth-mouse-not-available-after-pairing-as-input-device-on-slackware-current-4175712778/

Maybe we are supposed to show some dialog? (I remember my macbook did this everytime I connected wth it)

nilo85, 2024-02-03:
After I ran
hciconfig hci0 down
and sudo hciconfig hci0 up
it connected and works in UI

nilo85, 2024-02-03:
```
[ 3858.651048] systemd-journald[320]: /run/log/journal/64a7bca1d41cd476466b1933642b3dfe/system.journal: Journal header limits reached or header out-of-date, rotating.
[ 3863.485806] Bluetooth: hci0: security requested but not available
[ 3897.452660] Bluetooth: hci0: security requested but not available
[ 3931.446277] Bluetooth: hci0: security requested but not available
[ 3934.509698] Bluetooth: hci0: security requested but not available
[ 3965.444477] Bluetooth: hci0: security requested but not available
[ 3971.780692] Bluetooth: hci0: security requested but not available
[ 3973.153151] Bluetooth: hci0: security requested but not available
[ 3999.447873] Bluetooth: hci0: security requested but not available
[ 4033.478659] Bluetooth: hci0: security requested but not available
[ 4067.488741] Bluetooth: hci0: security requested but not available
[ 4086.705658] systemd-journald[320]: Data hash table of /run/log/journal/64a7bca1d41cd476466b1933642b3dfe/system.journal has a fill level at 75.1 (1537 of 2047 items, 524288 file size, 341 bytes per hash table item), suggesting rotation.
[ 4086.705692] systemd-journald[320]: /run/log/journal/64a7bca1d41cd476466b1933642b3dfe/system.journal: Journal header limits reached or header out-of-date, rotating.
[ 4095.474177] Bluetooth: hci0: unexpected SMP command 0x0b from f6:8b:ab:35:a8:f3
[ 4096.048526] input: StadiaNFNZ-a8f3 as /devices/virtual/misc/uhid/0005:18D1:9400.0004/input/input5
[ 4096.049065] hid-generic 0005:18D1:9400.0004: input,hidraw0: BLUETOOTH HID v1.00 Gamepad [StadiaNFNZ-a8f3] on dc:a6:32:23:68:2d
```

nilo85, 2024-02-03:
I removed the udev rules, rebooted.

Device was paired but non functional, then I did hciconfig hci0 down and up, and controller connects again, so I suspect there is really some sort of dance we are supposed to do with it that does not work automatically

nilo85, 2024-02-03:
PS, none of the buttons seem to be mapped as hotkey

nilo85, 2024-02-03:
Managed to figure out how to remap controller and get meny key working =)

However I guess we should figure out how to get that security request fixed properly.
I can try to figure out how to export a controller mapping tomorrow, auto worked good except for the menu button

nilo85, 2024-02-03:
Toggling bluetooth off and on via UI has the same effect.

So currently to get it working on Lakka you need to pair it, then Services > Bluetooth Off and then On again and it should tell you a Stadia controller was connected

nilo85, 2024-02-03:
Regarding binding there seems to be one already https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Google%20LLC%20Stadia%20Controller%20rev.%20A.cfg

But it gives a different name when connected over bluetooth maybe that is why it does not map it out of box ( had to rename it to whatever mine as called for Mupen64 on my mac...)

My generated config has these rows
```
input_driver = "udev"
input_device = "StadiaNFNZ-a8f3"
```

vs
```
input_driver = "udev"
input_device = "Google LLC Stadia Controller rev. A"
```

JayMcGann, 2024-08-14:
> I dont have a stadia controller unfortunately You can send the config here and will be added to the next lakka version https://github.com/libretro/retroarch-joypad-autoconfig

I am also having issues getting the stadia controller working in lakka

JayMcGann, 2024-08-14:
> Regarding binding there seems to be one already https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Google%20LLC%20Stadia%20Controller%20rev.%20A.cfg
> 
> But it gives a different name when connected over bluetooth maybe that is why it does not map it out of box ( had to rename it to whatever mine as called for Mupen64 on my mac...)
> 
> My generated config has these rows
> 
> ```
> input_driver = "udev"
> input_device = "StadiaNFNZ-a8f3"
> ```
> 
> vs
> 
> ```
> input_driver = "udev"
> input_device = "Google LLC Stadia Controller rev. A"
> ```
> 
> EDIT: Might also be they changed the device name when you flashed it to permanent BT mode and the existing bindings worked with pre-flashed controller over cable

nilo85 - did you ever get this working?

Fainought, 2024-08-14:
Never managed to get it to work, although it's been a while since I last tried to.

Marco Pignatelli

mtoensing, 2024-08-14:
I moved to Batocera because of this. It supports Stadia Wireless and connected via cable out of the box.

JayMcGann, 2024-08-14:
> I moved to Batocera because of this. It supports Stadia Wireless and connected via cable out of the box.

What is strange is that I was able to get it working with RetroArch inside Ubuntu?

JayMcGann, 2024-08-16:
I was able to get it to work with  SSH ( I think this is what worked??) :

agent on
default-agent
power on
pairable on
scan on
connect DF:7F:CD:B9:13:AF 
trust <DF:7F:CD:B9:12:AF> 

the button mapping often needs to be redone and the device id remains N/A