libretro/Lakka-LibreELEC #1826: Lakka 4.3 ODROID-C4 - WiFi Module 5BK - no wifi interface detected

mvrk69, 2023-02-02:
I'm using Lakka 4.3 on my odroid-c4, 
but wifi adapter (WiFi Module 5BK - https://www.hardkernel.com/shop/wifi-module-5bk) is not working, no wifi interface is detected.

Before was showing some problem with firmware missing, that was fixed on https://github.com/libretro/Lakka-LibreELEC/issues/1768)

But now, from what i can see seems the wifi driver is missing too:

[ 171.962995] usb 1-1.1: USB disconnect, device number 3
[ 178.120613] usb 1-1.1: new high-speed USB device number 5 using xhci-hcd
[ 178.273258] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[ 178.273290] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 178.273304] usb 1-1.1: Product: 802.11ac NIC
[ 178.273315] usb 1-1.1: Manufacturer: Realtek
[ 178.273325] usb 1-1.1: SerialNumber: 123456
[ 178.366249] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[ 178.367211] Bluetooth: hci0: RTL: rom_version status=0 version=1
[ 178.367232] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[ 178.367640] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_config.bin
[ 178.367942] Bluetooth: hci0: RTL: cfg_sz 10, total sz 31990
[ 178.819218] Bluetooth: hci0: RTL: fw version 0x829a7644

Lakka:~ # lsusb -tv
/: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
ID 1d6b:0003 Linux Foundation 3.0 root hub
|__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
ID 2109:0817 VIA Labs, Inc.
/: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
ID 1d6b:0002 Linux Foundation 2.0 root hub
|__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
ID 2109:2817 VIA Labs, Inc.
|__ Port 1: Dev 5, If 0, Class=Wireless, **Driver=btusb**, 480M
ID 0bda:c820 Realtek Semiconductor Corp.
|__ Port 1: Dev 5, If 1, Class=Wireless, **Driver=btusb**, 480M
ID 0bda:c820 Realtek Semiconductor Corp.
|__ Port 1: Dev 5, If 2, Class=Vendor Specific Class, **Driver=**, 480M
ID 0bda:c820 Realtek Semiconductor Corp.

by-gnome, 2023-02-05:
Please attach full `dmesg` log and `lsmod`.

mvrk69, 2023-02-06:
dmesg:
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/10669399/dmesg.txt)

lsmod:
[lsmod.txt](https://github.com/libretro/Lakka-LibreELEC/files/10669400/lsmod.txt)

mvrk69, 2024-04-24:
Still no go,

```
[    8.529701] Bluetooth: Core ver 2.22
[    8.529857] NET: Registered PF_BLUETOOTH protocol family
[    8.529873] Bluetooth: HCI device and connection manager initialized
[    8.532366] Bluetooth: HCI socket layer initialized
[    8.532400] Bluetooth: L2CAP socket layer initialized
[    8.532782] Bluetooth: SCO socket layer initialized
**[    8.580943] rtw_8821cu 1-1.1:1.2: Firmware version 24.11.0, H2C version 12**
**[    8.954924] rtw_8821cu 1-1.1:1.2: failed to download firmware**
**[    8.959301] rtw_8821cu 1-1.1:1.2: failed to setup chip efuse info**
**[    8.959328] rtw_8821cu 1-1.1:1.2: failed to setup chip information**
**[    8.971225] rtw_8821cu: probe of 1-1.1:1.2 failed with error -22**
[    8.971500] usbcore: registered new interface driver rtw_8821cu
[    8.992231] usbcore: registered new interface driver btusb
[    9.009904] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[    9.010912] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    9.010944] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_fw.bin
[    9.014318] Bluetooth: hci0: RTL: loading rtl_bt/rtl8821c_config.bin
[    9.014971] Bluetooth: hci0: RTL: cfg_sz 10, total sz 34926
[    9.504867] Bluetooth: hci0: RTL: fw version 0x75b8f098
[    9.734935] Bluetooth: MGMT ver 1.22
```

Ntemis, 2024-04-24:
you need this firmware
http://wtware.com/fw/mkzip.php?file=rtw88/rtw8821c_fw.bin

Ntemis, 2024-04-24:
https://github.com/home-assistant/operating-system/issues/3205

closed 2024-04-24 by Ntemis